ra3xdh/qucs_s #439: Add voltage references library

ra3xdh, 2024-01-08:
Add library containing SPICE models for voltage references ICs like TL431, AD581, REF191, etc.

ra3xdh, 2024-01-08:
[image]

tomhajjar, 2024-01-08:
There are a number of popular National Semi voltage/current references/regulators. LM334, LM385, LM4041, LM4050

Some of these devices can be substitutes for the other. Some come in fixed and adjustable versions. TL431 is more popular than all.

ra3xdh, 2024-01-08:
Yes, I am planning to add more devices. But I didn't find the model for LM385 yet.

tomhajjar, 2024-01-08:
It's an older device mostly replaced by the LM4041/4050 series.

ra3xdh, 2024-01-10:
I have uploaded a new library containing models that I have found. I may add more models if I found something. Closing this as completed.