ra3xdh/qucs_s #440: Schematic Editor: Edit Component Properties Remove Bug

3813127458, 2024-01-08:
"Edit Component Properties" -> "Remove" gives wrong result:

[RTF_N4_master.zip](https://github.com/ra3xdh/qucs_s/files/13865978/RTF_N4_master.zip)

[image]

tomhajjar, 2024-01-08:
Sorry I never mentioned this...

This happens on ALL the PARAM, Qucs Formula, Nutmeg Formula...etc forms. Doesn't matter what item/line you select/highlight, the bottom (or is it the top) item/line gets deleted when you click Remove.

The ONLY way I found to get around this is to use the Move UP button (or is it the Move Down button) and move EVERYTHING you DON'T want deleted UP (or is it DOWN) out of the way before you click Remove . 

[image]

ra3xdh, 2024-01-09:
Please test the latest commit on the `current` branch. I have added a fix for this.

3813127458, 2024-01-09:
So far it works fine on branch 24.1-6791ef07
I will test it further by using.

Would be nice if it is possible to select multiple items for removal by `Ctrl` + item + item ...
The `Add`, `Move Up` and `Move Down` buttons need to be locked in case multiple items are selected.
However, this behavior is of minor importance... could be skipped or implemented later.

closed 2024-01-16 by ra3xdh