ra3xdh/qucs_s #441: Templates for ngspice

tomhajjar, 2024-01-08:
I'm slowly making a "Template" folder with analysis examples including Nutmeg formulas needed to plot data.

I don't know the status of the S-parameter file capability so I didn't include an example. Last time, Qucs couldn't process s2p files with Mag, Angle formatted data.

[image]

[Templates_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/13867002/Templates_ngspice_prj.zip)

tomhajjar, 2024-01-22:
I've improved most of the examples in the project. Will be updating it this week.

ra3xdh, 2024-02-02:
I have uploaded Ngspice templates to `examples/templates_ngspice` subdirectory. Closing as completed. Thanks for the contribution!

closed 2024-02-02 by ra3xdh

iwbnwif, 2024-02-02:
Thank you, these templates are really useful. A few observations:
1. AC_TRAN_FOUR_OpAmp.sch has the "open data display after simulation" checked even though there are no charts on the .dpl page.
2. genericopa_sub.sch is missing file genericopa.cir
3. Pwr-Amp_wingspread_analysis.sch doesn't appear to have a 'dd' label, thus v(dd) is invalid?
4. S-parameter_file_analysis.sch is missing FL455kHz.s2p

ra3xdh, 2024-02-02:
> genericopa_sub.sch is missing file genericopa.cir

I have added missing file

> S-parameter_file_analysis.sch is missing FL455kHz.s2p

I excluded this schematic from examples pack. The attached s2p file was about 1Mb size. It's better avoid to keep such large files in git.

iwbnwif, 2024-02-02:
> "dd" is in the nutmeg equation.
Ah, sorry I see it now. Should the Nutmeg Simulation=SW1 instead of =dc?

tomhajjar, 2024-02-02:
@ra3xdh 

I redid the s-parameter file example to use the s-parameter file "137MHz_BPF.s2p". 

This is the same filter in S-parameter_passive_analysis.sch so is a better example.

On a side note, 137MHz_BPF.s2p was created by Qucs using qucsconv on the original LC filter. It's a feature Qucs-S doesn't have.

[image]

[Templates_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/14144046/Templates_ngspice_prj.zip)

iwbnwif, 2024-02-02:
@tomhajjar 
Shouldn't this be Simulation=SW1?
[image]

tomhajjar, 2024-02-02:
I created the schematic from the ngspice example, 100W.sp. I simplified the analysis to 8 Ohms.

Vadim wrote the Nutmeg equations and Parameter sweep entries and got it to work. See #405.

https://github.com/ra3xdh/qucs_s/issues/405