ra3xdh/qucs_s #442: Phantom nets

tomhajjar, 2024-01-08:
Attached is a schematic with a type of "Phantom" nets I talked about in #426. A "phantom" net can be caused by a deleting a cell connected to nets or a cut-paste or moving multiple items. 

In this case I deleted 2 OpAmps and tried to substitute new ones. This created 4 nets that move as if all 4 are selected when I try to replace the OpAmps. 

1st pic is schematic before trying to move OP1 into position. 2nd pic shows how all 4 nets move in sync with OP1 even though only OP1 was selected.

[image]

[image]

[AC_TRAN_Active_Filter_analysis.zip](https://github.com/ra3xdh/qucs_s/files/13867193/AC_TRAN_Active_Filter_analysis.zip)

wawuwo, 2024-01-14:
I've been trying to reproduce the issue using using two OpAmps like in the attached schematic and I haven't succeeded. Moving and deleting OpAmps in my synthetic example seems to work fine:
[image]

I also tried to take the attached schematic, delete dangling labels and put OpAmps in their places, add labels again. Then I moved and deleted OpAmps, copied and pasted them in a new place — and there was no problem with labels… But if the schematic is taken "as is", then moving OpAmps moves labels as well. It's strange. 

I've discovered a case when moving a component causes movement of a label:
[image]

It may be the same bug or a related one…  It may also be somehow related to an accumulated internal state of `Schematic` object. There is definitely a problem that should be looked further into.

>In this case I deleted 2 OpAmps in the original schematic and tried to substitute new ones.

@tomhajjar could you please provide the original schematic without dangling labels and describe step by step how to transition it into "having dangling labels" state?

tomhajjar, 2024-01-14:
I can't replicate the issue on my end. I don't know what sequence caused the nets to remain after deleting the two opamps and placing two back.

tomhajjar, 2024-01-16:
The nets need to be connected via wires to the component. 

If the net is connected directly to the component without a wire it will disappear if the component is deleted.

wawuwo, 2024-01-16:
> The nets need to be connected via wires to the component.
> 
> If the net is connected directly to the component without a wire it will disappear if the component is deleted.

Well, this seems to be alike to the bug with a resistor and a wire I showed in a previous reply. I'm working on the problem, but unfortunately the source code is not so easy to comprehend, so no quick results on the horizon :(

tomhajjar, 2024-01-16:
I would call this a very low priority "bug". It happens so rarely and when it does it's generally obvious and can be fixed by deleting the nets/wires and re-place them.

ra3xdh, 2024-06-02:
Closing because of unknown conditions to reproduce this issue. This is very rare schematic editing issue, which should not appear when using a normal schematic editing procedure. I have never observed floating 1x1 point nets in my schematics.

closed 2024-06-02 by ra3xdh