ra3xdh/qucs_s #446: AudioIC.lib Graphics

tomhajjar, 2024-01-09:
I did a quick clean-up of the new AudioIC.lib graphics. Pin labels were missing on 3 of the 4 parts. 

[image]

[AudioIC_prj.zip](https://github.com/ra3xdh/qucs_s/files/13874784/AudioIC_prj.zip)

ra3xdh, 2024-01-10:
I have just uploaded the fixed library. Thanks!

closed 2024-01-10 by ra3xdh

csrabak, 2024-01-20:
I've made a short "test circuit" per figure 9-1 _Typical Application_ of datasheet enclosed in this zip project.

Confront with figure 6-4, same doc, shows the model has a low frequency response different from the spec.

I've strived to avoid any impedance which could impart another zero in the freq resp and put the attacking voltage source directly connected to the input.

[image]

[LM386-test-ckt.zip](https://github.com/ra3xdh/qucs_s/files/13998720/LM386-test-ckt.zip)

tomhajjar, 2024-01-20:
The LM3886 is simulated over on the KiCAD site.

csrabak, 2024-01-20:
The IC I used is the LM386 and not the LM3886.

Which now I see as an opportunity for improvement: have these devices show the part id as other ICs do.

I surmise the origin of the model must be found and see if there is room for the betterment of the model, as it seems strange the IC being all DC coupled inside have a response which is below nominal gain up to 400 Hz by two decibels.

tomhajjar, 2024-01-20:
LM386 is very old and most would use proven circuits that have been around for decades.

I mentioned the LM3886 because it is popular with hobbyists and also in the Qucs-S  library. The LTspice and diyAudio forums would have the models hobbyists use. TI has a "transient" model for the LM3886.

For products like TVs, Manufacturers use class D amps with full digital interfaces. Analog audio power amps are slowly going away...

csrabak, 2024-01-21:
Yes I know of models of LM386, non official since 1995 By D. Dilatush.
Sverre, LA3ZA published his _disapointment_ with the model ten years ago.
Zabb Csaba also publish some Spice models, and his appraisal of the simulation is that the output in fig 6-4 would have been ploted for the pin 5 _before_ the decoupling capacitor, and in fact the spec for the measurement is not clear, given the figure has 50+ years (clearly is a copy and paste from earlier NS handbooks).  His solution for the reconciliation is to increase the DC decoupling cap C1 in the test circuit to one millifarad, a value our ancestors surely did no use even considering the IC was supposed to work with loads down to four ohms.
However, with this value the curve gets less "high pass" in the lower frequencies, so I think no further investment in this model is warranted.
WRT the LM3886 it is a higher output amplifier, whereas the LM386 is a low power amp which is used extensively in low power ham radio transceiver (and receiver) projects, thus my interest in having a workable model.
Thanks for the work you share with our community.

csrabak, 2024-01-21:
@tomhajjar would it be too cumbersome to include pin numbers in these symbols?

tomhajjar, 2024-01-21:
Since this program isn't used to lay out PCBs and pin numbers depend on the package chosen or number of devices in a package, adding them can cause problems. The pins are labeled with [pin function.

You are welcome to make your own versions of the devices as sub-circuits and assign pin numbers. You can also make your own libraries. The included libraries only have a small number of devices, so you will need to make your own sub-circuits

I made multiple schematics with symbols that you can use to make your own devices or libraries or edit existing libraries.

[Symbol_creation_prj.zip](https://github.com/ra3xdh/qucs_s/files/14003231/Symbol_creation_prj.zip)

csrabak, 2024-01-24:
I don't see any other place to comment on this, and please move to a wiki or similar, if warranted.

Testing the LM386 model with Xyce and Spiceopus in addition to Ngspice, which was tested by @tomhajjar, I needed to remove the parameter **Nk** from the transistor models otherwise the Spiceopus would choke and abort the simulation.