ra3xdh/qucs_s #447: Qucs-S misbehaving.... or is it ngspice....

tomhajjar, 2024-01-09:
This is in reference to #436. ngspice is unable to process Tlines as used as delay elements in TI "transient" Spice models.

Attached are 3 simple voltage delay line simulations using different Td parameters and Trans Sim times.

V_Delay.sch works

V_Delay_2.sch output contains strange discontinuities

V_Delay_3.sch creates a huge V_Delay_3.dat.ngspice file, causes Qucs-S to become unresponsive AND even after using Windows Task Manager to close and re-starting Qucs-S, Qucs-S starts hogging resources if any analysis is restarted. I was forced to reboot Windows. 

No idea what ngspice was doing. I will try to generate a netlist to test ngspice using DuSPice.

[image]

[image]

[image]

[Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/13879220/Test_prj.zip)

tomhajjar, 2024-01-10:
Any attempt to open V_Delay_3.sch causes Qucs-S to become unresponsive.  This is after a reboot...

[image]

ra3xdh, 2024-01-10:
> V_Delay_2.sch output contains strange discontinuities

I cannot reproduce the discontinuities on my machine. Everything works as expected. Please check your Ngspice version. The discontinuities are surely not Qucs-S error. 

[image]

> V_Delay_3.sch creates a huge file V_Delay_3.dat.ngspice, causing Qucs-S to become unresponsive

Ngspice may adjust the time step automatically. You have the simulation duration 5ms and rise time of the pulsed source 1ns. Ngspice may decide to reduce time step to 1ns at some conditions and you will get around 5000000 points in dataset. Qucs-S tries to convert this huge dataset and may freeze. The huge dataset is not a normal simulation operation. I can add an option to ask user if the large dataset is detected. But it is not obvious how to distinguish "large" and "normal" size datasets.

ra3xdh, 2024-01-10:
>  Is there a "setting" that is causing Qucs-S to "choke" on the large file?

The XML dataset format used by Qucs-S has numerous design issues and doesn't support large data volumes. The switching to HDF5 https://docs.hdfgroup.org/hdf5/develop/_intro_h_d_f5.html may help. The HDF5 format is designed to support both large small data files. But I don't consider this to be done in the near future because of time constraints.

tomhajjar, 2024-01-11:
I tested the same circuit V_Delay_2.sch in LTSpice and it works as expected. 

LTSpice netlist was exported and ngspice again generated data with slightly different discontinuities. 

Still waiting for response over at ngspice...

[image]

[image]

[Delay_line_netlists.zip](https://github.com/ra3xdh/qucs_s/files/13909132/Delay_line_netlists.zip)

ra3xdh, 2024-01-12:
I still cannot reproduce discontinuities using the Linux version. This may be something build specific. Closing the ticket as not related to Qucs-S GUI level.

closed 2024-01-12 by ra3xdh

tomhajjar, 2024-01-12:
Dietmar at ngspice has reproduced my data. Something changed between ngspice release 39 and 40.

https://sourceforge.net/p/ngspice/discussion/120973/thread/440b76e297/

[image]

[image]

tomhajjar, 2024-01-12:
I compared releases ngspice 42-64 with ngspice 38-64 and the data is dramatically different.

The data using ngspice 38-64 has slight differences compared with LTSpice.

[image]
[image]
[image]