ra3xdh/qucs_s #448: Instance description field tighter to symbol

dwarning, 2024-01-10:
I believe this was asked already but didn't find it.
The instance Verilog-A generated description field is very far from the referenced symbol in schematic.
[image]

What should I do that this field comes tighter to symbol?

ra3xdh, 2024-01-10:
Two solutions of your problem are possible:
 
* Move the device information text in the symbol editing mode (F9 on VA file or subcircuit). The existing schematics will be not updated automatically.
* Right click on the text and select "Move component text". The move the text block with mouse.

closed 2024-01-10 by ra3xdh

dwarning, 2024-01-10:
Sorry - first method did not success.
After F9:
[image]

Then I moved the Text:
[image]

Then I saved, reload modules and substitute the device in schematic:
[image]

Text is still far away.

Second method works, but I must repeat it every time.