ra3xdh/qucs_s #449: Additions to Voltage References Library

tomhajjar, 2024-01-11:
Per #439, I added a number of components to VoltageReferences.lib. LM334, LM4041_1.2 and LM4041_ADJ.

I used "REF" for the Prefix instead of "X". Is there a reason you chose X?

[image]

[image]

[Voltage_References_prj.zip](https://github.com/ra3xdh/qucs_s/files/13897104/Voltage_References_prj.zip)

ra3xdh, 2024-01-12:
> I used "REF" for the Prefix instead of "X". Is there a reason you chose X?

The "X" is a prefix for subcircuit in SPICE. That's why I am using it everywhere in libraries for ICs designation. You may also use REF as well, the device prefix in library doesn't affect simulation capabilities.

tomhajjar, 2024-01-12:
I understand but the "new" SpiceOpamp.lib used OP so I assumed you were moving away from using X.