ra3xdh/qucs_s #452: Add YLIN (s2p) device support for Xyce

ra3xdh, 2024-01-12:
### Discussed in https://github.com/ra3xdh/qucs_s/discussions/450

<div type='discussions-op-text'>

<sup>Originally posted by **michal777** January 12, 2024</sup>
In xyce reference guide (https://xyce.sandia.gov/files/xyce/Xyce_Reference_Guide_7.6.pdf) there's note about component:

_2.3.24. Linear device
The linear (YLIN) device allows an S-, Y-, or Z-parameter model to be used to define an N-port device. It
is mostly commonly used as part of a Harmonic Balance (HB) analysis_

Is it possible to do harmonic balance simulation with touchstone file in qucs-s? I can't find s-parameters file block in components when xyce is selected.</div>

tomhajjar, 2024-01-12:
Does the QucsXyceLib1.lib done by Mike Brinson in 2021 have any relevance? It has a source called Component port_device_sin.

[image]

[QucsXyceLib1.zip](https://github.com/ra3xdh/qucs_s/files/13919173/QucsXyceLib1.zip)

ra3xdh, 2024-01-12:
I have added the support for YLIN device. It will be available since the next release. It seems that Xyce supports only HB simulation for this device type. Closing as completed.

closed 2024-01-12 by ra3xdh

michal777, 2024-01-13:
Thanks,
I'm trying to test it. I'm not sure the order of nets is correct. "Ref" node of s2p or s3p is grounded but in the netlist there are no connections to ground.
[image]

ra3xdh, 2024-01-14:
I have fixed the pins order by the recent commit on `current` branch. Please update.