ra3xdh/qucs_s #454: How to build qucs_s on Windows?

NoClassDefError, 2024-01-13:
Hello, I'm doing some additional development based on qucs_s on Windows. 

After installing the flex and bison dependencies, I attempted to build the project using both the Visual Studio 2019 compiler and the MinGW64 8.1 compiler; however, both attempts resulted in errors. How can I build qucs_s on Windows? Is it necessary to modify the code?

[image]

ra3xdh, 2024-01-13:
VisualStudio build is not supported. The using of standalone MinGW installation in principle should work, but is not recommended. The best way is to install MSYS2 https://www.msys2.org/ It provides Unix shell and package manager on the Windows system. The additional CMake options are not required. @zergud is using Windows as the main development platform. Probably he can advice something.

NoClassDefError, 2024-01-14:
Thanks very much for your help!  I've attempted using MSYS2-mingw64 with multiple source versions, and unfortunately, I encountered a consistent compiling error indicating that a header file, q3scrollview.h, cannot be found, even though it does exist. The error message is provided below:
[image]

> VisualStudio build is not supported. The using of standalone MinGW installation in principle should work, but is not recommended. The best way is to install MSYS2 https://www.msys2.org/ It provides [...]

ra3xdh, 2024-01-14:
I cannot reproduce the mentioned build error. Everything works as expected on my MSYS2 installation. Are you sure you have set the proper Makefile format with "-G" option? Cmake on MSYS2 may require explicit setting of this option. I am using `cmake -G "MinGW Makefiles <source_dir>`

NoClassDefError, 2024-01-15:
Thanks for your help! By properly setting the cmake generator to MinGW64, I've successfully build and run the qucs-s!

closed 2024-01-15 by NoClassDefError