ra3xdh/qucs_s #456: XSpice Xanalogue.lib and examples

tomhajjar, 2024-01-13:
I "updated" Xanalogue.lib and it's test project XSPICE_Xanalogue_prj done by Mike Brinson in 2016. I cleaned up the symbols and text in Xanalogue.lib as well as the schematics in XSPICE_Xanalogue_prj. The new library Xanalogue.lib is in the project.

One test schematic Tran_XA_ilimit.sch generates errors that I don't understand.

Xanalogue.lib is a good alternative to Ideal.lib.  I haven't tested using both libraries in the same schematic.

[image]

[XSPICE_Xanalogue_prj.zip](https://github.com/ra3xdh/qucs_s/files/13928972/XSPICE_Xanalogue_prj.zip)

ra3xdh, 2024-01-30:
Thanks for the contribution. I have updated the library. Closing this as completed.

closed 2024-01-30 by ra3xdh

tomhajjar, 2024-09-28:
@ra3xdh 

I revisited the Xanalogue.lib and test schematics. With all the changes to ngspice, the schematics needed a refresh. However I'm having two issues.

1) "DC_XA_summers.sch" uses XA_summer_x2 and XA_summer_x3. Should Sum2=1 and Sum3=2?

2) No matter what values I enter into XA_ilimit_SE, I always get an "CLIMIT function linear range less than zero"

Lastly all the devices with "DE" outputs were originally tested with only the "+" output. The "-" output was connected to ground. I changed the schematics to resistor loads on + and -. The devices don't work unless both outputs have a "load".

[image]
[image]

[XSPICE_Xanalogue_prj.zip](https://github.com/user-attachments/files/17171927/XSPICE_Xanalogue_prj.zip)