ra3xdh/qucs_s #458: conda install qucs?

joamatab, 2024-01-14:
How can we make a conda-forge package of qucs?

We are exploring [Schematic driven layout ](https://gdsfactory.github.io/gplugins/notebooks/20_schematic_driven_layout.html) for [gdsfactory](https://gdsfactory.github.io/gdsfactory/) using Qucs 

@proppy
@nikosavola

ra3xdh, 2024-02-12:
Hello, I have a little interest in maintaining conda package for Qucs-S myself. But I will accept if someone will provide a package. Please submit necessary PRs if required.

ra3xdh, 2024-02-12:
> Qucs

Also what exactly package do you need? Qucs-S, Qucs or Qucsator?

ra3xdh, 2024-03-20:
Still be no response on this. I am unable to maintain conda package myself. Closing this. May be reopened if someone will take maintenance on conda package.

closed 2024-03-20 by ra3xdh