ra3xdh/qucs_s #460: Transformers.lib error

tomhajjar, 2024-01-15:
I guess I didn't test Transformers.lib well enough. LossyTransformer2 is missing the braces around Rp and Rs. 

I will post a fix soon.

[image]

[image]

ra3xdh, 2024-01-15:
I have uploaded the `Transformer.lib` provided in #367 to the repository in the last week. The demo package that I shared with you doesn't have this version. Do you have a newer version than posted in #367?

tomhajjar, 2024-01-15:
To be safe I would use the Transformers.lib attached above. 

I checked what I uploaded a while ago and it didn't match what came with Qucs-S 24.1.0...

tomhajjar, 2024-01-15:
Attached are 4 projects containing various test circuits for Transformers.lib and transformer sub-circuits used to create Transformers.lib. They should be very useful for new users and speed learning.

[Transformer_projects.zip](https://github.com/ra3xdh/qucs_s/files/13942816/Transformer_projects.zip)

ra3xdh, 2024-01-30:
The library version attached in the first message and the library in the repository differs only by variable registry. The library update is not needed, because Ngspice is not registry dependent. Closing this.

closed 2024-01-30 by ra3xdh