ra3xdh/qucs_s #461: Transformer Netlist and Library wrong

tomhajjar, 2024-01-15:
During ngspice 42plus testing I found an issue with an old project that replicated the work done by Mike Brinson to create Transformers.lib. My versions of the PositiveCouplingPS and NegativeCouplingPS transformers don't work anymore. Qucs flagged "Warning: Unusual leading characters".

The Netlist and Library I created are full of "=" characters. I cannot see what caused this or why the simulation data is wrong.

[image]
[image]
[image]
[image]
[image]

[Transformer_sub_Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/13941474/Transformer_sub_Test_prj.zip)

ra3xdh, 2024-01-15:
The inductors are containing the `=` symbols on the continuation lines. You have to delete these symbols manually and the simulation error will be gone. I don't know why the `=` symbols appeared here. I never observed such issues on my schematics containing the red inductors and cannot reproduce this. 
[image]

tomhajjar, 2024-01-15:
I tried to open the schematics with Qucs-S 0.0.24 and subcircuit symbols won't load. I don't remember this happening before.

[image]

[image]

closed 2024-01-16 by tomhajjar