ra3xdh/qucs_s #463: Proper use of FOUR and FFT with AC and Trans

tomhajjar, 2024-01-16:
I cannot get AC, Trans and FFT data in the same analysis while AC, Trans and FOUR works. 

Compare C300_FFT to C300_FOUR plots:

1. For C300_FFT, ac.v(in), ac.v(out) and frequency are listed twice. "frequency" has 301 and 8193 data points. How is Qucs storing the AC versus FFT data? 

2. Why does FFT use the same start/stop as the AC analysis? FOUR doesn't have this restriction. I can set the start and stop.

3. The FFT plot is garbage unless I delete the AC analysis. 

[image]

[image]

[Amp_prj.zip](https://github.com/ra3xdh/qucs_s/files/13944020/Amp_prj.zip)

ivandi69, 2024-01-16:
> ac.v(in), ac.v(out) and frequency are listed twice. "frequency" has 301 and 8193 data points. How is Qucs discerning the AC versus FFT data?

I stumbled upon that issue too. 
Pull request https://github.com/ra3xdh/qucs_s/issues/463 fixes it for custom simulations.

tomhajjar, 2024-01-16:
Pull request #463?

Attached is ngspice netlist of 100 watt audio amp from the ngspice examples folder with simultaneous AC, Trans and FFT.

It works and the FFT looks good. I would like to be able to do this with Qucs-S.

[image]

[100w_amp_netlists.zip](https://github.com/ra3xdh/qucs_s/files/13945110/100w_amp_netlists.zip)

ivandi69, 2024-01-16:
> Pull request https://github.com/ra3xdh/qucs_s/issues/463?

Sorry, pull request https://github.com/ra3xdh/qucs_s/pull/412.

dwarning, 2024-01-16:
Only for information - not an answer, sorry: The difference to capabilities of other popular simulators is that ngspice VDMOS model has true self-heating built-in. So you can investigate how the impact is on distortion if the cooling chain is insufficient. E.g. look for the both thermal nodes of the power mosfets in transient analysis.

ra3xdh, 2024-01-16:
I am aware about this problem. The usage of FFT and AC in the same time is not allowed. @ivandi69  PR #412 has effect only for script simulation type and doesn't help to resolve this issue. The reason of the conflict of AC and FFT is that both are using `frequency` as X-axis variable. I cannot figure out how to resolve this conflict. Probably the setting of different plot name for AC and FFT may help. This should be done in `Ngspice::createNetlist()`

tomhajjar, 2024-01-16:
The source for DuSpice is available. Maybe it will reveal how Holger does it. The analysis % done is also in the code. 

https://ngspice.sourceforge.io/experimental/ng_gui_sources.7z

tomhajjar, 2024-01-17:
I did more testing. The Nutmeg script required to do AC, Trans and FFT is very simple. In addition Trans and FFT can use different tran statements allowing independent analysis. 

As you know inside Qucs-S, FFT frequency "span" is "locked" to the AC analysis start/stop frequency and FFT frequency "resolution" and the Transient analysis are "restricted" to the same Tran statement. These limitations don't exist in ngspice and DuSpice and KiCAD don't have them.

My attempt using a Nutmeg script inside Qucs-S was a failure...

[image]

[image]
[image]
[image]

[FFT_test_prj.zip](https://github.com/ra3xdh/qucs_s/files/13958609/FFT_test_prj.zip)

ra3xdh, 2024-01-17:
@tomhajjar This could be recommended only as the workaround. It's need to find a way to resolve conflict of the X-axis variable name for AC and FFT. This should be fixed in C++ level but not in documentation level.

tomhajjar, 2024-01-17:
Wouldn't the source for DuSpice reveal how Holger did it?

ra3xdh, 2024-01-17:
The sources of DuSpice will not help. DuSpice doesn't convert schematic to netlist. It only provides an interface for the console Ngspice. You can run Ngspice in terminal and obtain the same result as using DuSpice. `Plot` and `write` uses different concept. Probably the setting of the plot name will help (chapter 17.3 of the Ngspice manual). Nutmeg script syntax is not obvious and I cannot to figure out how to resolve the X-axis variable name conflict.

tomhajjar, 2024-01-17:
I'm confused... 

I tested OpAmp_ac_trans_fft.sch and generated a netlist -> OpAmp_ac_trans_fft.qucs

Edited OpAmp_ac_trans_fft.qucs by adding plot statements and deleting destroy all and reset creating -> OpAmp_ac_trans_fft.sp.

Used DuSpice on OpAmp_ac_trans_fft.sp and the data files OpAmp_ac_trans_fft_XX.txt and plots are correct.

If the data files are labeled "ac", "fft" and "tran" and "frequency" and # of points is different for the ac and fft txt files, why are the ac and fft plots BOTH using "frequency" and # of points from ac.txt when fft should be using frequency and # of points from fft.txt?

[image]
[image]
[image]

[AC_Tran_FFT.zip](https://github.com/ra3xdh/qucs_s/files/13965201/AC_Tran_FFT.zip)

ra3xdh, 2024-01-17:
> Isn't the problem that the Qucs-S FFT plot is using the amplitude data from OpAmp_ac_trans_fft_fft.txt but the frequency data from OpAmp_ac_trans_fft_ac.txt? 

All SPICE datasets are assembled into one Qucs XML dataset. If there exists two X-variables with the same name, it will be conflict. It is not a straightforward task to resolve this issue. I don't expect this to be fixed by the next release.

tomhajjar, 2024-01-18:
I created a test project. It compares data files from Qucs-S and DuSpice, ngspice. I had DuSpice generate data in two file formats.

One can compare the Qucs file "OpAmp_ac_trans_fft.dat.ngspice" to DuSpice files "OpAmp_ac_trans_fft_fft.txt" and "OpAmp_ac_trans_fft_fft.text".

I cut out the fft data from "OpAmp_ac_trans_fft.dat.ngspice" and created "OpAmp_ac_trans_fft_fft.qucs" to make comparison easier.

The data set size is the same "Points: 129" but the Qucs file is missing "half" of the vector data. I don't know the ngspice file format so I'm unsure if this "missing" data is redundant and irrelevant for plotting or displaying data in Qucs.

Duspice, ngspice
-2.692580420695830e-04,0.000000000000000e+00
1.198932402578705e-04,0.000000000000000e+00
Qucs
-2.692580421893e-04+j0.000000000000e+00

[ATF_file_prj.zip](https://github.com/ra3xdh/qucs_s/files/13979179/ATF_file_prj.zip)

tomhajjar, 2024-01-19:
> Only for information - not an answer, sorry: The difference to capabilities of other popular simulators is that ngspice VDMOS model has true self-heating built-in. So you can investigate how the [...]

@dwarning
Do you know the original source for the ngspice example 100W.sp? It's in the same vmos folder as 100W_wingspread.sp?

I found a number of issues with it. Gain and Phase plots are wrong and the use of "fourier 1K V(out)" doesn't seem to do anything. Deleting it doesn't effect the plots. I recreated the schematic in LTspice and Qucs-S.

I assume due to the extra source V3, 100W.sp is trying to use the Tian method.

[Amp_100W_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/13991749/Amp_100W_ngspice_prj.zip)

ra3xdh, 2024-02-01:
Fixed via #513

PR #513 (2024-02-01): Ngspice netlist rework, fixes #463
This rework fixes #463 using the recently added custom prefix feature.
Prefixing is activated only if conflicting simulations are found, so **the existing schematics are not affected**.

Every simulation has its own output file named spice4qucs.<simulation_name>.extension, ex: spice4qucs.ac1.plot.
If any conflicting simulations are found, ex: AC and FFT, prefixing is activated and when the dataset is built the <simulation_name> prefix applies to all simulations. If a conflicting simulation is added to a schematic the existing diagrams have to be updated with  [... cut]