ra3xdh/qucs_s #465: Integrate Qucs-RFLayout with Qucs-S

ra3xdh, 2024-01-16:
There exist a project called Qucs-RFLayout, which is capable to convert Qucs schematic containing microstrip devices into RF layout and OpenEMS simulation script. It seems this project is abandoned now. The latest commit was 2 years ago. It's need to evaluate its status and add integration with Qucs-S. @thomaslepoix?

tomhajjar, 2024-01-16:
The accuracy of the dimensions are only as good as the models. Is there a source of accurate models that have been proven?

thomaslepoix, 2024-02-20:
(I just answered in the other thread)

@tomhajjar I have no idea about the mathematical models Qucsator uses and their accuracy.

Qucs-RFlayout operates purely on geometrical level, the dimensions used are the ones written by the user. Then I based the shapes on what I found in the documentation of AWR and ADS, this kind of stuff :
[image]

I shipped how shapes are actually drawn as few pdf files within binary packages (`share/doc/qucsrflayout/`)

ra3xdh, 2024-02-20:
@thomaslepoix Thanks for the answer. I propose a following integration of Qucs-RF-layout and Qucs-S. I can add a menu entry in the tools menu and automatize an execution of the CLI qucs-rflayout tool. It's need to provide an instruction for the user how to install the tool.

zergud, 2024-02-20:
@thomaslepoix do you plan to further develop Qucs-RF-layout?
if no, IMHO, the best solution would be to include this tool in qucs-s codebase (with attribution, of course)

this will simplify support, development and there will be no need to compile binaries for different platforms

ra3xdh, 2024-02-20:
> do you plan to further develop Qucs-RF-layout?

@zergud The questions about the future of Qucs-RFLayout is answered here: https://github.com/thomaslepoix/Qucs-RFlayout/issues/32#issuecomment-1953298537

thomaslepoix, 2024-02-23:
Ok, so I will do some renovation work soon

@ra3xdh Perfect. For distributions with a supported package manager, I assume installation instructions will be as simple as adding my repository for the user, and adding to Qucs-S an optional dependency link (like `Recommends` or `Suggest` on both APT and RPM) to Qucs-RFlayout. I currently don't support MacOS and FreeBSD, I assume it won't be hard to add.

Regarding Windows, I am not really satisfied with the all-in-one zip I provide, I would rather provide an installer or something like that. How do you manage that?

Also did some changes occur in the schematic file format since Qucs?

ra3xdh, 2024-03-20:
@thomaslepoix 

I did some testing of QucsRFLayout recently and it works fine with Qucs-S schematics. It would be sufficient to replace `qucs` by `qucs-s` in the source code to switch netlist converter to Qucs-S. Please be aware of #639 if you would like test Qucs-S CLI operation. Use the version from Git that has fixed #639 

> Also did some changes occur in the schematic file format since Qucs?

The most of changes of schematic format are related to diagrams extensions and new SPICE devices. The diagrams are not exported to netlist. The microstrip-related stuff is not changed. The Qucsator netlist format is not changed too.

ra3xdh, 2024-03-20:
Basic integration of Qucs-RFLayout tool has been added by #648

closed 2024-03-21 by ra3xdh

thomaslepoix, 2024-03-23:
@ra3xdh Great, thank you !

I tried on `current` and can confirm it works fine if I either create a netlist from the CLI  `qucs-s -i test/lpf/lpf.sch -n -o test/lpf/lpf.net`, either run `qucs-s -i test/lpf/lpf.sch`, then go `Tools > RF Layout`.

I have spotted a strange behavior though : 

If I just launch Qucs-S from the same directory as above and open the schematic through `File > Open`, then `Tools > RF Layout`, a popup says

```txt
Netlist error

Failed to open file: /home/<user>/.qucs/lpf_openems.s2p
```

And it produces a netlist that does not include actual nets (so RF Layout computes garbage), below are the two netlist versions. I precise in both cases the `.s2p` files are located just next to the schematic at `./test/lpf/<file>` instead of `~/.qucs/<file>`, but from the CLI it doesn't seem to matter.

- from `qucs-s -i test/lpf/lpf.sch` :

```txt
# Qucs 24.2.0  /home/<user>/Documents/Qucs-RFlayout/test/lpf/lpf.sch

Pac:P1 _net0 gnd Num="1" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
MLIN:MS1 _net0 _net1 Subst="Sub1" W="500 um" L="4.5 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS8 _net1 _net2 Subst="Sub1" W1="500 um" W2="20.5 mm" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS2 _net2 _net3 Subst="Sub1" W="20.5 mm" L="1.5 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS9 _net3 _net4 Subst="Sub1" W1="20.5 mm" W2="500 um" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS3 _net4 _net5 Subst="Sub1" W="500 um" L="18 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS10 _net5 _net6 Subst="Sub1" W1="500 um" W2="20.5 mm" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS4 _net6 _net7 Subst="Sub1" W="20.5 mm" L="2.5 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS11 _net7 _net8 Subst="Sub1" W1="20.5 mm" W2="500 um" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS5 _net8 _net9 Subst="Sub1" W="500 um" L="18 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS12 _net9 _net10 Subst="Sub1" W1="500 um" W2="20.5 mm" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS6 _net10 _net11 Subst="Sub1" W="20.5 mm" L="1.5 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS13 _net11 _net12 Subst="Sub1" W1="20.5 mm" W2="500 um" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS7 _net12 _net13 Subst="Sub1" W="500 um" L="4.5 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
Pac:P2 _net13 gnd Num="2" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
SUBST:Sub1 er="4.2" h="800 um" t="35 um" tand="0.035" rho="1e-10" D="0"
.SP:SP1 Type="lin" Start="20 MHz" Stop="5 GHz" Points="499" Noise="no" NoiseIP="1" NoiseOP="2" saveCVs="no" saveAll="no"
Eqn:Eqn1 Qucs_S11="dB(S[1,1])" Qucs_S21="dB(S[2,1])" Export="yes"
Pac:P3 _net14 gnd Num="3" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
Pac:P6 _net15 gnd Num="6" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
Pac:P4 _net16 gnd Num="4" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
Pac:P5 _net17 gnd Num="5" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
Eqn:Eqn2 OpenEMS_S11="dB(S[3,3])" OpenEMS_S21="dB(S[4,3])" Measure_S11="dB(S[5,5])" Measure_S21="dB(S[6,5])" Export="yes"
SPfile:X1 _net14 _net16 gnd File="{/home/<user>/.qucs/lpf_openems.s2p}" Data="rectangular" Interpolator="linear" duringDC="open"
SPfile:X2 _net17 _net15 gnd File="{/home/<user>/.qucs/lpf_measure.s2p}" Data="rectangular" Interpolator="linear" duringDC="open"
```

- from `qucs-s` + `File > Open` :

```txt
* Qucs 24.2.0  /home/<user>/Documents/Qucs-RFlayout/test/lpf/lpf.sch

Pac:P1  gnd Num="1" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
MLIN:MS1   Subst="Sub1" W="500 um" L="4.5 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS8   Subst="Sub1" W1="500 um" W2="20.5 mm" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS2   Subst="Sub1" W="20.5 mm" L="1.5 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS9   Subst="Sub1" W1="20.5 mm" W2="500 um" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS3   Subst="Sub1" W="500 um" L="18 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS10   Subst="Sub1" W1="500 um" W2="20.5 mm" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS4   Subst="Sub1" W="20.5 mm" L="2.5 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS11   Subst="Sub1" W1="20.5 mm" W2="500 um" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS5   Subst="Sub1" W="500 um" L="18 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS12   Subst="Sub1" W1="500 um" W2="20.5 mm" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS6   Subst="Sub1" W="20.5 mm" L="1.5 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
MSTEP:MS13   Subst="Sub1" W1="20.5 mm" W2="500 um" MSModel="Hammerstad" MSDispModel="Kirschning"
MLIN:MS7   Subst="Sub1" W="500 um" L="4.5 mm" Model="Hammerstad" DispModel="Kirschning" Temp="26.85"
Pac:P2  gnd Num="2" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
SUBST:Sub1 er="4.2" h="800 um" t="35 um" tand="0.035" rho="1e-10" D="0"
.SP:SP1 Type="lin" Start="20 MHz" Stop="5 GHz" Points="499" Noise="no" NoiseIP="1" NoiseOP="2" saveCVs="no" saveAll="no"
Eqn:Eqn1 Qucs_S11="dB(S[1,1])" Qucs_S21="dB(S[2,1])" Export="yes"
Pac:P3  gnd Num="3" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
Pac:P6  gnd Num="6" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
Pac:P4  gnd Num="4" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
Pac:P5  gnd Num="5" Z="50 Ohm" P="0 dBm" f="1 GHz" Temp="26.85"
Eqn:Eqn2 OpenEMS_S11="dB(S[3,3])" OpenEMS_S21="dB(S[4,3])" Measure_S11="dB(S[5,5])" Measure_S21="dB(S[6,5])" Export="yes"
SPfile:X1   gnd File="{/home/<user>/.qucs/lpf_openems.s2p}" Data="rectangular" Interpolator="linear" duringDC="open"
SPfile:X2   gnd File="{/home/<user>/.qucs/lpf_measure.s2p}" Data="rectangular" Interpolator="linear" duringDC="open"
```

ra3xdh, 2024-03-23:
>  it works fine if I either create a netlist from the CLI 

The Qucs-S should create netlist automatically from schematic and substitute result to RFLayout. No manual invocation of `qucs` from CLI is needed. 

> If I just launch Qucs-S from the same directory as above and open the schematic through File > Open, then Tools > RF Layout, a popup says

It's strange. I have just created a test schematic containing both microstrips and s2p device and everything works fine. Qucs-S accepts both absolute and relative (inside projects) paths and RFLayout tools generates layout. There may be problems when opening schematic that belongs to the project outside project #567

ra3xdh, 2024-03-23:
In addition to above, always use absolute paths in subcircuits and s2p devices when you are working with standalone schematic that doesn't belong to the project.

ra3xdh, 2024-03-23:
I have tried the `lpf.sch` from the `test` subdirectory of RFLayout source tree. The simulation doesn't work unless you substitute absolute paths in s2p blocks. The Qucs-0.0.20 shows the same error. So, this behavior was not intriduced in Qucs-S. The implementation of #640 may resolve the issue with file paths.

tomhajjar, 2024-03-23:
In Windows lpf.sch runs as-is without modification.

The version of Qucs-S I have doesn't have the menu selection for RFLayout

[image]

thomaslepoix, 2024-03-23:
> @thomaslepoix are you going to release a new version soon? I need a Windows version for testing.

@tomhajjar This integration does not require any change on the Qucs-RFlayout side, so I think the old [2.0.0 package](https://github.com/thomaslepoix/Qucs-RFlayout/releases/download/2.0.0/qucsrflayout-2.0.0-x86_64.zip) will be fine as all dependencies are bundled within it and nothing has happened from then feature-wise.

But yes, I am currently reworking the way I cross compile for Windows, new release ASAP.

Thanks for feedback on `lpf.sch`.

@ra3xdh Interesting, I was not really aware of so called projects beyond `DataSet` or
`DataDisplay` properties. Let's see if #640 fixes that.

thomaslepoix, 2024-04-06:
Just to say I implemented compatibility with Qucs-S for when Qucs-RFlayout is used in standalone mode : trying to call both Qucs and Qucs-S by default for netlist generation.