ra3xdh/qucs_s #466: .spiceinit

tomhajjar, 2024-01-17:
Adding ".spiceinit" to your schematic doesn't seem to do anything. The .spiceinit file gets written to the spice4qucs folder which AFAIK, ngspice never sees.

For Windows, putting a .spiceinit file into the "user" folder will override any .spiceinit placed anywhere else. ngspice will only use the  .spiceinit file in the "user" folder. 

I've tried using the ngspice command "unset ngbehavior" to no avail. I've been hashing it out with Holger. 

No idea what happens in Linux but doubtful ngspice looks in the spice4qucs folder.

[image]

[image]

ra3xdh, 2024-01-17:
I cannot reproduce the reported issue on Linux machine. Everything works as expected. I am using spiceinit regularly in my simulations and everything has effect. Need to check this on Linux. Ngspice for linux should always prefer the spiceinit in the working directory to the default Ngspice. Maybe Ngspice doesn't like the space in the working directory path. Need to test using Windows.

ra3xdh, 2024-01-17:
I didn't change the spiceinit processing since v0.0.24. So, no new bug should be introduced in the recent releases.

tomhajjar, 2024-01-17:
This came up using KiCAD and DuSpice. Some examples wouldn't work unless I deleted the .spiceinit from my user folder. 

Using the "unset" command hasn't worked yet. I need to try more variations...

https://sourceforge.net/p/ngspice/discussion/120973/thread/afefdb21be/

tomhajjar, 2024-01-17:
Holger says ngspice searches for the .spiceinit file in the following sequence.

* directory named in env variable SPICE_USERINIT_DIR
* current directory
* HOME directory
* USERPROFILE directory

I have not figured out where the first two are. Do you tell ngspice to use "spice4qucs"? ngspice doesn't seem to look at the "current" director  for the .spiceinit file or where the netlist is.

I can confirm that ngspice looks for the .spiceinit file in the C:\Ngspice-xx_64\bin and User\ directories

I'm doing most of my testing using KiCAD and DuSpice. I need to test Qucs-S more.

tomhajjar, 2024-01-17:
.spiceinit file contains "set ngbehavior=ltpsa"

If I copy spiceinit to my User directory, circuit works.

If I enable the .spiceinit on the schematic, circuit fails.

If I copy spiceinit to C:\Ngspice-xx_64\bin, circuit fails

If I copy spiceinit to "spice4qucs", the file is deleted or overwritten by Qucs-S.

[image]

[image]

[spiceinit_test_prj.zip](https://github.com/ra3xdh/qucs_s/files/13968788/spiceinit_test_prj.zip)

tomhajjar, 2024-01-18:
If you look at the news for release 38 one gets the feeling things might have changed:

 + Search for .spiceinit firstly in a user defined directory named in env
      SPICE_USERINIT_DIR, then in the current directory, then in HOME, then
      in USERPROFILE.

https://github.com/imr/ngspice/blob/master/NEWS#L15

tomhajjar, 2024-01-18:
I discovered my .spiceinit on my schematic had "set ngbehavior=ltps" instead of "set ngbehavior=ltpsa". It fooled me into thinking I had a Qucs issue. I didn't realize Qucs deletes/overwrites the .spiceinit file in the spice4qucs folder. I would copy .spiceinit into the folder and Qucs was deleting it during a run...again fooling me... :-(

Having .spiceinit in my User folder is OK for Qucs but an issue for some ngspice examples which won't work...

closed 2024-01-18 by tomhajjar