ra3xdh/qucs_s #468: Phantom wires while movineg seleted components by mouse or arrow keys

zergud, 2024-01-19:
wire and node appears instead of decrease wire length
length of phantom wire equals movement distance - dx for horizontal, dy for vertical

[image]

zergud, 2024-02-18:
@wawuwo this bug may be intresting for you, while you plunged into the depths of schematic )
it appear while wire segment moving in wire bend direction

for example - if we move such selection UP

[image]

wawuwo, 2024-02-19:
@zergud that's an interesting case, I'd call the bug what happens when you move selected components down :)

I mean, having selected a _whole_ wire, you expect it move as a whole, as one piece. And after such move, it lies on a node and automatically connected with it, creating two wires. The result is logically equivalent to the picture on the right side:
[image]

So every move up after the first one is actually an attempt to lay one wire on top of another and overlapping segment is optimised away, that's why it doesn't grow further up. This looks perfectly reasonable.

But! When you move the whole wire _down_, it is extended, but not moved down as one piece as one would think. 
But! We can look at this "extension" like this: at first the wire is moved down as one piece and to keep connected to the node a short wire is inserted. And then these two wires are optimised into one wire.

zergud, 2024-02-19:
I already broke my head once and decided to take a break ))
try to turnoff snap to grid and move... nice picture )

but "the truth is somewhere near"

ra3xdh, 2024-02-19:
> I already broke my head once and decided to take a break ))

The wires with one hanging end don't affect the simulation. This issue may be left as is if not possible to fix. The bug was presented since Qt4 version.

zergud, 2024-02-19:
> try to turnoff snap to grid and move
ups, i mean "move with ctrl pressed"

>The bug was presented since Qt4 version.
this doesn't mean that we don't have to fix it ))

wawuwo, 2024-02-20:
@zergud
>ups, i mean "move with ctrl pressed"

I still can't get what is the bug you mean :( Here I move a wire with the Ctrl pressed:
[image]

Is it possible for you to record a video?

zergud, 2024-02-20:
hmmm... I can't repeat ))
many phantom pins appears while moving without snap to grid
probably it fixed by #509 or elsewhere )

wawuwo, 2024-02-28:
@zergud, shouldn't we close the issue or maybe mark it as "backlog" or something? I mean no one managed to reproduce the issue…

zergud, 2024-02-28:
[image]

wawuwo, 2024-02-28:
>do you have Windows PC?

Unfortunately no, I don't.

Regarding the example you've provided: I finally see what you consider a bug, but I can't get why it is a bug for you :) I mean let's take this example:

https://github.com/ra3xdh/qucs_s/assets/40355531/185a0209-db01-4b72-b5cb-8d62e77ddcde

It looks to me like absolutely reasonable and expected behaviour: the whole wire is selected, moved as _one piece_ and placed at new location. In your example the same action is performed but with a set of components instead, not a wire alone…

zergud, 2024-02-28:
if you select many wires then you can get a lot of unconnected nodes
it seems I'm the only one who has problems with this behavior 
lets close issue and forget
thanx )

closed 2024-02-28 by zergud