ra3xdh/qucs_s #469: 2.1.0 GUI is overscaled on 4k displays (Win 11)

iwbnwif, 2024-01-19:
All elements of the Qucs-S GUI are too big when using Windows 11 display scaling.

This is a new issue in 2.1.0 and didn't occur in 2.0.0.

It looks as though the scaling factor is being applied twice. I have the scale set to 150% and everything is approximately 1.5 times bigger than it should be (i.e., compared to 2.0.0 or other Qucs's.

It is possible to work around the problem by creating the following VBS script:

```
Dim WshShell, WshEnv
Set WshShell = CreateObject("WScript.Shell")
Set WshEnv = WshShell.Environment("PROCESS")
WshEnv.Item("QT_SCALE_FACTOR") = "0.67"
WshShell.Run "qucs-s.exe
```"

The GUI size is fine in Ubuntu on the same hardware and with the same desktop scaling.

ra3xdh, 2024-01-20:
Thanks for reporting and providing a workaround. I have never observed scaling issues with fullHD hardware using Linux or Windows platform. Unfortunately I have no access to 4k hardware to test what happens. Probably some scaling options should be applied as the `QApplication` attribute. The best way if someone will provide a patch fixing this issue.

zergud, 2024-01-22:
this is quite a complex problem and relates to  non-integer scale factors (such as 150%)
we have different behavior for qt5 and qt6

if set Qt::HighDpiScaleFactorRoundingPolicy::Floor in version 2.1 we obtain correct GUI scaling (menu toolbar)
but wrong font scaling in schematic (problem exists in earlier versions as well)

need futher investigation!

[image]

iwbnwif, 2024-01-23:
I notice that the Windows binary ('portable package') for 2.1.0 is built using Qt5. I am not able to build on Windows at the moment, but would be happy to test a Qt6 version if there is a binary available.

wawuwo, 2024-02-28:
This ticket has been silent lately… Is GUI still overscaled on 4k in 24.1.0?

iwbnwif, 2024-02-28:
Yes, the binary available from the main site still has the overscaling problem.

zergud, 2024-02-28:
You need to add option - line 832
and test on 4k display

[image]

iwbnwif, 2024-02-29:
Unfortunately, I am unable to build on Windows at the moment. Previously @ra3xdh made binary available - is it possible to do that again?

iwbnwif, 2024-04-19:
Just to let you know, 24.2.1 is still overscaled.

Is it possible to add setHighDpiScaleFactorRoundingPolicy(Qt::HighDpiScaleFactorRoundingPolicy::Floor) or setHighDpiScaleFactorRoundingPolicy(Qt::HighDpiScaleFactorRoundingPolicy::RoundPreferFloor) in the next Windows build?

I don't think this will have any detrimental affect on non-4K displays.

zergud, 2024-04-19:
Some "detrimental affect on non-4K display" appears when set scale 1.5 for example.. menu items overlaps etc.

ra3xdh, 2024-04-20:
Maybe switching Windows build to Qt6 will resolve this problem? I am considering this option since the next release.

iwbnwif, 2024-04-20:
> Some "detrimental affect on non-4K display" appears when set scale 1.5 for example.. menu items overlaps etc.

I see - I tried 1.25 on non-4k and it seemed to be okay. Perhaps the best thing would be to include it as an optional setting?

I will try to setup a build environment on Windows so I can better test this and create a PR.

iwbnwif, 2024-04-20:
I can confirm that setHighDpiScaleFactorRoundingPolicy(Qt::HighDpiScaleFactorRoundingPolicy::Floor)  works on Windows 11 4k displays with 150% scale factor.

There are some small artefacts, for example the simulation tool drop down on the toolbar is too small to fit "Qucsator" and the Thickness value in the Edit Diagram properties box only shows half a number. Nonetheless, the application is much more useable than without the setting.

ra3xdh, 2024-05-23:
I have tested the Qt6 build using the Windows-11 and found no scaling issue. I have used FullHD laptop display with 125% scaling applied. Need to test this with #723 which imporoves the schematic rendering. I think this issue will be self-fixed after swithching Windows build to Qt6 by default and merging #723.

iwbnwif, 2024-05-28:
Thank you, the scaling is much better now.

I have tested with the continuous build snapshot for 0faf4ce. The only thing that is noticeably not quite right is the spacing between the top level menus.

[image]

But this is a really minor point and I think this issue can be closed now.

ra3xdh, 2024-05-30:
Closing as completed. The Windows build will be switched to Qt6 since the next release.

closed 2024-05-30 by ra3xdh