ra3xdh/qucs_s #470: Netlist format reasoning

tomhajjar, 2024-01-20:
For help in debugging, Holger and Dietmar generally want a netlist compatible with ngspice. To use the netlist exported by Qucs-S, I have to edit it to be usable. Is there a reason for the following entries I always delete/modify/comment-out before I send the netlist to ngspice using DuSpice?

This line causes an ngspice error.
.INCLUDE "C:/QUCS-S 24.1.0/share/qucs-s/xspice_cmlib/include/ngspice_mathfunc.inc"

These generate unneeded files I always delete or comment-out so as not to flood the ngspice\bin folder
echo "" > spice4qucs.cir.noise
echo "" > spice4qucs.cir.pz

These won't allow plot statements I add to the file to work/display.
destroy all
reset
exit

I always comment-out (*) the write statements and add plot statements using only the variables I want plotted.

csrabak, 2024-01-20:
I've been there, done that and I think that if the format with those "ornaments" are needed for the way qucs-s calls the ngspice simulator, then we've to create a script in one of the languages _du jour_ for this task (I'm thinking AWK, but perhaps Python would be the choice) and have it as an accessory.

just my .019999...

ra3xdh, 2024-01-20:
The netlist format is not subject of discussion and not subject of change. The netlist generated by Qucs-S is not indended to be run by a normal user (not developer) except the debug purposes. 

> This line causes an ngspice error.
.INCLUDE "C:/QUCS-S 24.1.0/share/qucs-s/xspice_cmlib/include/ngspice_mathfunc.inc"

This was already asked and answered several times before. This line contains definitions needed for compact modeling support (functions like `limexp`). This line may be removed if the schematic doesn't use export to Verilog-A.  

> These generate unneeded files I always delete or comment-out so as not to clutter the ngspice\bin folder.

These two files are needed for PZ simulation. I am rewriting them every time to ensure that there is no remains of the old simulation.

> I usually comment-out (*) the write statements and add plot statements using only the variables I want plotted. 

The `plot` has no effect when used in a headless mode from within child process. The `write` is required to collect datasets that are later converted into Qucs XML dataset.

closed 2024-01-20 by ra3xdh