ra3xdh/qucs_s #471: Qucs-s 2.1.0 doesn't render utf-8 characters correctly anymore

csrabak, 2024-01-20:
In windows 10, I've schematics since version 1.0.2 which employed local characters be it in the text boxes in the schematics, be it in the legends of the plots (saved in the .plt files).

When I open these projects with present 2.1.0 the i18n characters are rendered as a diamond with an interrogation in it!

Is it a regression bug or a new setting that has to be tuned?

tomhajjar, 2024-01-20:
I'm not having any issues with Win 10. Can you post a schematic and a screen shot of the issue?

ra3xdh, 2024-01-20:
I cannot reproduce this on Linux and Windows platforms. I have just created schematic in v1.0.1 and opened in the latest development build. The cyrillic UTF-8 letters are loader correctly. Could you post the `*.sch` file example with the broken symbols?

> the plots (saved in the .plt files).

Qucs-S doesn't produce `*.plt` files. Did you mean `*.dpl`?

[image]

csrabak, 2024-01-20:
Yes I did mean `.dpl` 
[image]
[mfb-bpf.zip](https://github.com/ra3xdh/qucs_s/files/13998251/mfb-bpf.zip)

Win 10 Single Lang Home Ed. Ver 22H2 OS compilation 19045.3930 Windows Feature Experience Pack 1000.19053.1000.0

zergud, 2024-01-20:
need more investigation: what font do you have in settings?
I changed the text in your file and everything is displayed correctly

[image]

csrabak, 2024-01-20:
FWIW I've never bothered to change the defaults for the fonts, be it in Windows or Linux.

[image]

Thanks for looking at it.

zergud, 2024-01-20:
for some reason your file is saved in ANSI encoding
I checked that all new dpl files are saved in utf-8 on my PC
:thinking:
[image]

tomhajjar, 2024-01-20:
Will this program help?

https://github.com/amrali-eg/EncodingChecker

[image]

csrabak, 2024-01-21:
@zergud thanks for finding the suspect!

I cannot understand how that was done, as I am not a hacker of these tools at all, OTOH now the issue has been pinpointed, I think be it manually be it by using the tool @tomhajjar point to, I can go "repairing" the projects on a needed basis.

Thanks to all, as it seems at the moment this is not something epidemic I'm closing the issue.

closed 2024-01-21 by csrabak