ra3xdh/qucs_s #472: Distortion example source format

tomhajjar, 2024-01-20:
I'm not having luck getting an old example working. Was the "special" source used for "Distortion simulation" removed requiring using a different source with text input format?

https://qucs-s-help.readthedocs.io/en/latest/ASim.html?highlight=distortion

[image]

[2N2222_dist.zip](https://github.com/ra3xdh/qucs_s/files/13996963/2N2222_dist.zip)

tomhajjar, 2024-01-20:
Think I got it. Not sure why source changed versus older versions of Qucs..

https://sourceforge.net/p/ngspice/discussion/ngspice-tips/thread/e58621f6/

ra3xdh, 2024-01-20:
> Not sure why source changed versus older versions of Qucs-S.

This source symbol was deprecated as redundant. It looks confusing and may mislead users. Use the red SPICE voltage source instead.

>  don't understand the input format "DISTOF1 0.2 DC 0 SIN(0 0.2 4k)".

Ngspice manual chapter 4.1 explains these parameters. From the Ngspice manual

> DISTOF1 and DISTOF2 are the keywords that specify that the independent source has distortion inputs at the frequencies F1 and F2 respectively (see the description of the .DISTO control line). The keywords may be followed by an optional magnitude and phase. The default values of the magnitude and phase are 1.0 and 0.0 respectively.

closed 2024-01-20 by ra3xdh