ra3xdh/qucs_s #479: CSV export mixes locale convention with English radix mark convention

csrabak, 2024-01-21:
I've exported the response of a circuit and the CSV file shows a mix of the locale, in this machine CSV files as per Microsoft convention use semicolon ';' to separate cells and the regional settings use the comma instead of point for the radix mark.

In the excerpt I enclose here it can be seen that Qucs-s exported the data using the locale for the cell separation but kept the numbers as it were in English locale (I think the functions that generate the strings used the C locale):

`"frequency";"r ngspice/ac.ganho_db";"i ngspice/ac.ganho_db"` 
```100;8.37328;0
100.077;8.39689;0
100.154;8.4205;0
100.231;8.44411;0
100.307;8.46771;0```

This was in Windows 10 single language home edition.

ra3xdh, 2024-01-26:
This is done intentionally. The CSV files are used for data exchange not only with Excel but also with Ocatve/Matlab and Python. Octave accepts only dot as the decimal delimiter.

closed 2024-01-26 by ra3xdh

csrabak, 2024-01-26:
I see. I suggest we put some place in future documentation update.