ra3xdh/qucs_s #480: Nutmeg equations using Fourier results of LM386

tomhajjar, 2024-01-21:
I can't figure out how to plot in dB the Four(v(out)). 

If I drive the LM386 into distortion, the dB FFT plot, S_dB=db(v(out)) clearly shows the distortion while the linear plot of Four(v(out)) does not. 

[image]
[image]
[image]
[image]

[LM386.zip](https://github.com/ra3xdh/qucs_s/files/14002043/LM386.zip)

csrabak, 2024-01-21:
Just for the sake of dividing and conquering, have you tried a minimum example directly in ngspice?

I'm lost about some results of it alone itself (including the one already reported by you @tomhajjar )

tomhajjar, 2024-01-21:
Unfortunately out of the 600+ examples provided by the ngspice team only one uses Fourier and I believe it was used in error, 100W.cir

ra3xdh, 2024-02-01:
#517 should resolve this problem.

ra3xdh, 2024-02-01:
My comment above was wrong. The attaching of Nutmeg equations is not allowed to FOUR simulation. The variables are only printed in the console or exported to the text file. Ngspice manual recommends to use FFT if the postprocessing with Nutmeg is needed.

closed 2024-02-01 by ra3xdh

tomhajjar, 2024-02-08:
Vadim can you show how to use Nutmeg to get Four in dB?

Using the "Log left axis grid" setting blanks the screen

[AudioIC_prj.zip](https://github.com/ra3xdh/qucs_s/files/14214000/AudioIC_prj.zip)

ra3xdh, 2024-02-09:
> Vadim can you show how to use Nutmeg to get Four in dB?

It is not possible by design. The FOUR results could be only printed in terminal. Ngspice manual recommends to use Spectrum analysis (FFT) instead.

csrabak, 2024-02-19:
Am I doing something absolutely wrong running this example with ver 24.1?

I made two different schematics to circumvent the issue of not allowing FFT and AC in same.

When I run the AC one, the delay calculation doesn't get the older results:

[image]

tomhajjar, 2024-02-19:
You should attach your files so others can help.

HB9GKW, 2025-01-27:
I was struggling with the same need of plotting fourier output in dB. As a workaround you could change y axis settings to logarithmic left axis grid and change units to dB. Doing this may result in "invalid data" if fourier results are small negative numbers. Injecting a tiny DC level (1nV) via a choke was for me the solution to keep things running.