ra3xdh/qucs_s #481: Suggestion to improve GUI

csrabak, 2024-01-21:
As I'm an old fart that started using computers when even in primitive unices (my first Unix machine was a Coherent Unix, which Mark Williams sold the X part as an add on, and it was to come by at that time in this corner of World), I am more used to keyboard commands a.k.a. _shortcuts_ or in the absence of them I recourse directly to the menues.

However, as I joined in a project where we assembled a small team of volunteers and in this post Covid time we started to convene to work my coleagues mentioned something could be a suggestion if it is not hard to put in the GUI:

The commands of the positioning menu, specially the aligment ones are in other tools put in a set of buttons which became almost _lingua franca_ for the operations as align by the top, or bottom, etc.

If they could be accommodate in qucs-s it would make it more intuitive for the people used to this interface.

zergud, 2024-01-21:
In #368 planned to implement shortcut manager for ALL Qucs-S actions, not only for those who are now has shortcut

csrabak, 2024-01-21:
Thanks @zergud !

I'll leave up to you if this should be already closed or wait #368

zergud, 2024-01-22:
will be fixed in #368

closed 2024-01-22 by zergud