ra3xdh/qucs_s #483: Qucs EDD project altered to work with Qucs-S doesn't work anymore

tomhajjar, 2024-01-21:
Vadim a Qucs MESFET example using an EDD that you fixed in the following thread doesn't work anymore.
https://github.com/ra3xdh/qucs_s/issues/152

File works under Qucs-S 0.0.24 and ngspice 38 but fails under Qucs-S 24.0.0 and ngspice 38, 42, 42plus. 
File also doesn't work under Qucs-S 0.0.24 with any ngspice rev >38.

The netlist has a slight variation between Qucs-S 0.0.24 and Qucs-S 24.0.0. 

The errors using Qucs-S 24.0.0 and ngspice 38 are different than ngspice 42/42plus. 

I started this because MESFETs.lib doesn't work and I wanted to fix it.  

[image]
[image]

[MESFET_Curtice_prj.zip](https://github.com/ra3xdh/qucs_s/files/14002604/MESFET_Curtice_prj.zip)

tomhajjar, 2024-01-22:
I think MESFETs.lib doesn't work on my PCs because my User profile name is "Tom Hajjar". 

The space " " is causing these errors running Qucs-S 0.0.24:

Error: Library C:\Users\Tom couldn't be loaded!
Error: Library Hajjar\.qucs couldn't be loaded!
Error: Library - couldn't be loaded!
Error: Library S\spice4qucs\qucs_cmlib\qucs_xspice.cm couldn't be loaded!

tomhajjar, 2024-01-22:
Using Qucs-S 24.1.0, I get different errors depending on the version of ngspice. 

ngspice 42plus generates an odd line in the log, "spinit in C:\Spice64_KLU\share\ngspice\scripts"

[image]

ra3xdh, 2024-02-17:
I cannot reproduce the attached error logs. The MESFET models work on my machine as expected. Moving this issue to discussion: https://github.com/ra3xdh/qucs_s/discussions