ra3xdh/qucs_s #484: qucs-s exports to log.txt file "ngspice-40 done" while installed ver is 42

csrabak, 2024-01-21:
In my windows 10 environment where I've not nsgspice (in fact c:\spice64\bin) in the path, nor any other programm which could be confused with,  whose test is in the command line issue `where ngspice_con` which comes empty.

However, in the 2.1.0 installation, after a simulation the line `ngspice-40 done` appears in the log.txt.

Is it spected behaviour?

Also in the properties modal window it seems there is not anymore an entry for ngspice location (I surmised when I saw at first, it could be because ngspice is finicky about its installation directory):

[image]

tomhajjar, 2024-01-22:
Here is my Windows configuration. It is user configurable so the specific folders are your choice. This can change in the future.

[image]

[image]

ra3xdh, 2024-01-22:
> the line ngspice-40 done appears in the log.txt

I have no such line in the source code. Qucs-S doesn't check the Ngspcie version and doesn't report it. Please check you settings and shell variables.

csrabak, 2024-01-22:
I've just checked and learned that qucs-s 2.1.0 comes with a ngspice executable (in fact two, including ngspice_con.exe) in its own bin directory[^1].

[image]

It is the ver 40 one!

I might have missed the new feature as the installation document (pointed to  by @tomhajjar) doesn't mention it and for the portable instructs to download and install the spice flavour.

[^1]: Similar technology is being employed by Kicad which ships with a DLL.

closed 2024-01-22 by csrabak