ra3xdh/qucs_s #487: Schematic Editor: Crtl+C plus Ctrl+V (Copy & Paste) of Circuit Blocks Error

3813127458, 2024-01-23:
Schematic editor: the copy & paste procedure of circuit blocks does not work seamless.

Two issues:
1) double placement after paste click
2) component names are not assigned to be compatible with the simulator netlist; Rx, Lx etc
    (net names I did not check)

[copy_and_paste.webm](https://github.com/ra3xdh/qucs_s/assets/75557552/9210b305-da86-43fe-bd7b-768d32593c26)

[HBM_S.zip](https://github.com/ra3xdh/qucs_s/files/14019900/HBM_S.zip)

Branch: 24.1-0d63ab95

Operating System: openSUSE Tumbleweed 20240121
KDE Plasma Version: 5.27.10
KDE Frameworks Version: 5.114.0
Qt Version: 5.15.12
Kernel Version: 6.6.11-1-default (64-bit)
Graphics Platform: X11

locale
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=

ra3xdh, 2024-01-23:
I cannot reproduce the copy-paste error using the same environment. Qt5.15+X11+KDE. Everything works as expected on my machine. I have tested Qt6 version too. Please check you system mouse settings. 

> component designators are not assigned to be compatible with the simulator netlist; Rx, Lx

This is done intentionally. Component names on schematic and in the SPICE netlist may be different.

3813127458, 2024-01-23:
Obviously it depends on the desktop general behaviour setting:

I prefer single click desktop actions (= "Opens them") in order to avoid double mouse clicks in general.
In case activating "Selects them" it work as expected. Could you reproduce it with "Opens them" setting?

Issue 2): component designator assignment should be done automatically (default) and manual (optional) in order to avoid huge rework of the circuit block to enable simulation. It could be very time consuming to assign manually the component designators in case of high number of copy & paste circuit elements. 

[image]

wawuwo, 2024-01-23:
@3813127458 
>double placement after paste click

I tried to reproduce using the app built from latest commit and that's what I've found: when a user presses `Ctrl-V` the app pastes copied elements at the current cursor position and switches to a sort of "insert mode" in which every mouse click pastes a new copy of elements until the user disables the mode by pressing `Esc`. See [copy.webm](https://github.com/ra3xdh/qucs_s/assets/40355531/cd12493e-9cb5-48d0-9894-1f2d8a4790b2)

You can see it in your video too: after the first copy appeared the "paste" button in the toolbar became "pressed down" 
[image]. And then I believe you clicked, which gave you a second copy.

To ensure that we're talking about the same app behavior, could you please check that each subsequent mouse click pastes a new copy of elements? i.e. you can do not a only a double placement, but a triple placement and so on too?

I don't know if it is a bug or a feature actually :)

ra3xdh, 2024-01-23:
>  the app pastes copied elements at the current cursor position and switches to a sort of "insert mode" i

This behavior was always present in Qucs since the very early Qt3 versions and never caused problems. Don't change it.

zergud, 2024-01-23:
I'll fix it

wawuwo, 2024-01-23:
@ra3xdh it's the `7ffa97f2` commit which introduces this bug.  I've built from the previous one (`457e7ac3`) and it's ok

closed 2024-01-23 by zergud

3813127458, 2024-01-23:
I can confirm that copy & paste now works perfect on recent branch 24.1-d3d80a00.

PR #489 (2024-01-23): Revert mousemove realization for paste
fixes #487
commit fd08e0ba (2024-01-23): revert mousemove realization for paste
fixes #487