ra3xdh/qucs_s #490: No libraries install by default

tomasApo, 2024-01-23:
Hi I have a fresh install from this qucs repository but it is not coming with any system libraries. Although I do have components through the components tab. 
[image]
[image]

**Any ideas or ways to solve this?**

tomhajjar, 2024-01-24:
What file did you use? Link?

https://ra3xdh.github.io/

ra3xdh, 2024-01-24:
@tomasApo The both RPM/DEB packages distributed from the OBS repository contain libraries. Use this link for install https://software.opensuse.org/download.html?project=home%3Ara3xdh&package=qucs-s WIndows installer also contain the libraries. I cannot reproduce the missing libraries issue n a fresh installation.

If you are using the package from Fedora main repository, you must install the libraries yourself: https://packages.fedoraproject.org/pkgs/qucs-s/qucs-s/ The Fedora team keeps libraries in a separate package: https://packages.fedoraproject.org/pkgs/qucs-s/qucs-s-library

ra3xdh, 2024-01-29:
No further information provided. Assume this was Fedora package installation that requires a separate library package. Closing this ticket.

closed 2024-01-29 by ra3xdh