ra3xdh/qucs_s #492: Statusbar improvements

zergud, 2024-01-24:
1. show diagram value under mouse poiner
2. remove simulator information from statusbar - duplicate simulator combobox value for now

ra3xdh, 2024-01-24:
> show diagram value under mouse poiner

Yes, this will implement some sort of cursor measurement. Probably it's need to consider to make it better visible for users. 

> remove simulator information from statusbar - duplicate simulator combobox value for now

Yes, the information on the status bar became redundant after introduction of the simulator combo box.

zergud, 2024-01-24:
look for PR #493 

[image]

one thing left: take aware for engineering/scientific for Axis

tomhajjar, 2024-01-24:
Will the 0,0 "Datum" AND location of the mouse cursor as indicated in the status bar ever be accurate?

The Datum and object locations as indicated by the mouse cursor in the status bar varies by Zoom and other metrics.

[image]

[Qucs_Datum.zip](https://github.com/ra3xdh/qucs_s/files/14039463/Qucs_Datum.zip)

zergud, 2024-01-24:
@tomhajjar yes, cursor position in status shows wrong value
but, this is not relevant to this issue
I suggest opening a new issue for this
p.s. is it really userful to get information about cursor position?

iwbnwif, 2024-01-24:
> p.s. is it really useful to get information about cursor position?
Not really. It could be shown for debug builds only, but it is also simple enough to use qDebug.

tomhajjar, 2024-01-24:
When making symbols it can be useful. Also some schematics I've downloaded are placed very far from the Datum it's difficult to find the "0,0" Datum cross (+). Maybe if the + was slightly larger it wouldn't be so hard to find.

zergud, 2024-01-25:
@tomhajjar fixed!

closed 2024-01-25 by ra3xdh