ra3xdh/qucs_s #494: More contributions for Portuguese translation

csrabak, 2024-01-24:
In menu Simulation:

[image]

Please update the following: 

1. "Tune" to _Ajustes_.
2. "Save netlist" to _Salvar netlist_.
3. "Simulator Settings" to _Ajustes dos Simuladores_.

In menu  Simulation->Simulator Settings. . .:

[image]

1. All "Select..." to _Selecionar..._.
2. "Setup simulators executable location" to _Configurar localização dos executáveis dos simuladores_.
3. "SPICE settings" to _Configurações do SPICE_.
4. "Ngspice executable location" to _Localização do executável ngspice_.
5. "Xyce executable location" to _Localização do executável Xyce_.
6. "SpiceOpus executable location" to _Localização do executável SpiceOpus_.
7. "Directory to store netlist and simulator output" to _Diretório para salvar a netlist e saída do simulador_.
8. "Extra simulator parameters" to _Parâmetros extras do simulador_
9. "Qucsator settings" to _Configurações do quscator_.
10. "Qucsator executable location" to _Localização do executável qucsator_.
11. "Apply changes" to _Aplicar mudanças_.

zergud, 2024-01-24:
there are many unfinished string for Portuguese translation
you can try to translate yourself without any knowledge of C++ or git
1. download standalone qtlinquist https://github.com/thurask/Qt-Linguist/releases/tag/20231206
2. download zip with source code from github
3. run linguist.exe and open translations/qucs_pt_PT.ts
4. translate everething whatever you want :) manual - https://doc.qt.io/qt-5/linguist-translators.html
5. send result file

or, if it is very unclear and difficult, please translate maximum number of strings from Qucs GUI at once and post in this issue 

[image]

wawuwo, 2024-02-11:
@csrabak Hi! What's set status of this issue? Are you still interested in translation? Or maybe you're working on it?

csrabak, 2024-02-17:
I'm interested and working on in, family issues have put me out of work and other endeavours, right now things are normalizing and I should finish this and send to you folks.

ra3xdh, 2025-01-03:
> I should finish this and send to you folks.

No activity after almost a year. Closing this.

closed 2025-01-03 by ra3xdh