ra3xdh/qucs_s #496: Text document (TextDoc) font is too small on HiDPI displays

iwbnwif, 2024-01-24:
TextDoc is the built in Qucs-S editor and it has a fixed font size that is set as the schematic font size - 1.

A reasonable setting for the schematic font size (that is in proportion to the symbols) is 10pt, but this creates a "Courier New" 9pt font for the text in TextDoc.

Is it worthwhile to add another option in Application Settings for the TextDoc font size? 

It may also be nice to allow mouse wheel zooming in TextDoc via QPlainTextEdit::zoomIn and QPlainTextEdit::zoomOut as a workaround for any edge cases or accessibility.

ra3xdh, 2024-02-10:
I have found one issue with this PR. If the text editor is already opened, the font size is not updated if the settings dialog is executed and `OK` pressed. It's need to close editor tab and open it again to update the font size. Is this expected behavior?

iwbnwif, 2024-02-10:
Yes, actually I followed the other font settings in the Applications Settings dialog and wrote "... (set after reload)" next to the setting.

I think this is only a minor inconvenience because I wouldn't expect users to change the default setting that often, but we could search through the open tabs to see if there is a text document and refresh it?

ra3xdh, 2024-02-10:
> I think this is only a minor inconvenience 

Yes, the things may be left as is.

PR #541 (2024-02-10): Add default TextDoc font to settings dialog
This change adds an extra entry to the settings dialog to allow the user to choose a default font for any documents opened in the text editor. The initial default value follows the system setting for a monospaced font.

I wasn't going to do this until after the settings refactoring in #533, but on the other hand it would be nice to close #496 before releasing 24.1.0.
closed 2024-02-10 by ra3xdh