ra3xdh/qucs_s #497: Assertion error when starting a new schematic

iwbnwif, 2024-01-24:
This issue relates to the development branch, e.g., statusbar-improvements, but is probably due to the MVC refactoring of the schematic layout.

Steps to reproduce:
1. Open Qucs-S
2. Without panning, scrolling, or zooming the blank main window, drag a lumped element onto the middle of the schematic
3. An assert occurs in Schematic::renderModel (line 2745 of the mentioned branch).

iwbnwif, 2024-01-25:
I am using Qt5. I can try again with Qt6 and report back. 

I should also add that this is on Ubuntu 22.04 standard Gnome desktop.

ra3xdh, 2024-01-25:
I cannot reproduce it using Qt5. I have tried the `current` branch and PR #493

zergud, 2024-01-25:
i thing must be <=
[image]
@wawuwo what do you think?

iwbnwif, 2024-01-25:
It is the first line that asserts, but after further testing it only seems to happen with larger window sizes.

The values for modelPoint are xp=955, yp=685 and for newMode are x1=0, y1=0, x2=799, y2=799.

iwbnwif, 2024-01-25:
It is the first line that asserts, but after further testing it only seems to happen with larger window sizes (I am using a 4k display).

The values for modelPoint are x()=955, y()=685 and for newModel are left()=0, right()=799. So it is the x() < right() test that is failing.

ra3xdh, 2024-01-25:
> I am using a 4k display

This explains why I was not able to reproduce it. I have tested only FullHD hardware.

wawuwo, 2024-01-25:
> It is the first line that asserts, but after further testing it only seems to happen with larger window sizes (I am using a 4k display)

This happens because when schematic is created its model size is set to 800x800 points (see [here](https://github.com/ra3xdh/qucs_s/blob/00ad40e2b224e6155ae90efeb5e4ccc340d2f529/qucs/schematic.cpp#L78)). If you place an element closer to top-left corner then assertion won't fail.

The hardcoded size was there before #444. When schematic is created its model size _should be_ equal to viewport size, because at the beginning we're displaying the schematic at 1:1 scale and it should occupy the whole viewport. When I was doing #444 I tried to set `ViewX2` and `ViewY2` to `viewport()->width()` and `viewport()->height()` accordingly, but unfortunately this didn't give good results. It seems that by the time `Schematic` constructor is called its widget doesn't have the final size and I received some strange values. Thus I decided to leave it untouched.

This kind of behavior hasn't been observed earlier because there were no assertions before #444. I think assertions should remain in place because they control the invariants.

Possible solutions are:
1. Assign `ViewX2` and `ViewY2` some very large values in `Schematic` constructor. Anyway this values are adjusted every time user scrolls or uses "Show all", so eventually they will become consistent.
2. Find a way to assign correct viewport size values to `ViewX2` and `ViewY2.

wawuwo, 2024-01-25:
You can assign the issue on me, I'll try to come up with a fix as soon as possible

iwbnwif, 2024-01-25:
Thank you for looking into this. 

As you say, a good work around for now is to quickly zoom in and out a new schematic before placing any components.

iolpltuisciaoln, 2024-01-25:
i have the same issue
```
qucs_s/qucs/schematic.cpp:2703: double Schematic::renderModel(double, QRect, QPoint, QPoint): Assertion `modelPoint.y() >= newModel.top() && modelPoint.y() <= newModel.bottom()' failed.
[1]    4097476 IOT instruction (core dumped)  /usr/bin/bin/qucs-s
```
```
commit 00ad40e2b224e6155ae90efeb5e4ccc340d2f529 (grafted, HEAD -> current, origin/current, origin/HEAD)
Date:   Wed Jan 24 08:51:11 2024 +0300

Update PT translation #482
```

PR #500 (2024-01-27): Fix assertion fails in 'renderModel'
This is an attempt to fix #497.

Using the build from this branch I tested the following cases on new empty schematic and also on schematic from a file.
1. Place an element in the middle of the view
2. The same as 1 for every corner
3. Zoom in and zoom out
4. Scroll and pan

I haven't observed any oddities or crashes.
closed 2024-01-27 by ra3xdh