ra3xdh/qucs_s #498: ngspice VDMOS electro-thermal model using Qucs-S

tomhajjar, 2024-01-25:
@dwarning pointed out ngspice offers a VDMOS "thermal" model while LTspice doesn't. I'm trying to get in to work using Qucs-S. 

I'm having issues and unsure if it's just me or a limitation in Qucs-S. I haven't figured out how to get a netlist as shown in the following pics.

[image]
[image]
[image]
[image]

[VDMOS_thermal_prj.zip](https://github.com/ra3xdh/qucs_s/files/14047768/VDMOS_thermal_prj.zip)

tomhajjar, 2024-01-26:
I've made some limited progress but only by hand editing the netlist and using DuSpice/ngspice. 

[image]

[image]

[image]

[VDMOS_thermal_prj.zip](https://github.com/ra3xdh/qucs_s/files/14059673/VDMOS_thermal_prj.zip)

dwarning, 2024-01-26:
"v(Tj)={Temp}" looks for me more as a misunderstanding. To set device temperature you should use instance parameter temp or dtemp.
Node Tj is more for sensing junction temperature.

tomhajjar, 2024-01-26:
So the KiCAD example Dual-NMOS-amp.cir is "technically" wrong because it uses: .ic v(Tj1)={envtemp} v(Tj2)={envtemp}

While the KiCAD example mosamp2.cir and ngspice example 100W.sp are correct.

All the examples simulated.

My problem was two fold. I tried to assign Tj and did it wrong...

[image]

[ngspice_thermal.zip](https://github.com/ra3xdh/qucs_s/files/14068047/ngspice_thermal.zip)

tomhajjar, 2024-01-27:
Holger pointed out I was running the device too hot and had too high a thermal Z to the heatsink and this was causing my problems. I made changes to Vgs max Rc and the simulations look OK. Once a solution to adding VDMOS thermal is found we should be good. 

Still haven't figured out how to overlay my plots...

[VDMOS_thermal_prj.zip](https://github.com/ra3xdh/qucs_s/files/14073476/VDMOS_thermal_prj.zip)

tomhajjar, 2024-01-28:
The above simulation used a built-in ngspice device, "VDMOS thermal".

For a GaN simulation where a thermal model exists, Holger created an a symbol, Holger_GaN_T.lib and the simulation called the subcircuit GaN_Pspice_GS66508T_L3V2.lib.  

https://ngspice.sourceforge.io/el-th-images/GaN-test.7z

I will try the same method in Qucs-S.

ra3xdh, 2024-02-02:
You need to give the `thermal` parameter to the SPICE generic device inside the subcircuit. This device supports passing arbitrary parameter string. After applying these modifications everything works as expected. See the attached screenshots.
[image]
[image]

closed 2024-02-02 by ra3xdh

tomhajjar, 2024-02-02:
I will confirm when I get new version. This could be of interest to @dwarning

tomhajjar, 2024-02-08:
With the latest nightly release I was able to finish a number of projects based on the ngspice example 100W.sp. 

The ngspice example does AC, Tran, Four, FFT, thermal and noise analysis. 

The Qucs-S version separates the analysis into AC/Tran/Four/FFT and Tran/thermal.

[image]

[image]

[image]

[Amp_100W_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/14214788/Amp_100W_ngspice_prj.zip)
[Amp_100W_thermal_prj.zip](https://github.com/ra3xdh/qucs_s/files/14214776/Amp_100W_thermal_prj.zip)

dwarning, 2024-02-08:
Thank you for the examples.
Unfortunately 100W_tran_thermal.sch didn't work. It comes with NaN for the thermal limiting function.
I have no problem with 100W.sp - but no success with the attached netlist.
 
[netlist.cir.txt](https://github.com/ra3xdh/qucs_s/files/14215282/netlist.cir.txt)

tomhajjar, 2024-02-08:
You have to edit the netlist to find the libraries and generate plots. Also need to comment out all the commands that delete the plots. You can also comment out the statements that generate unneeded files. 

[image]

[100W_thermal_prj.zip](https://github.com/ra3xdh/qucs_s/files/14215371/100W_thermal_prj.zip)

dwarning, 2024-02-08:
I must correct me: The problem is not compatibility mode, it is "set skywaterpdk" which was left over from other experiments.

tomhajjar, 2024-02-08:
@dwarning 

If ngspice would work with more companies spice models, thermal analysis would be more common.,

https://sourceforge.net/p/ngspice/discussion/120973/thread/4721495ffa/
https://sourceforge.net/p/ngspice/discussion/120973/thread/b18b28cdab/