ra3xdh/qucs_s #501: How could we pull the operational amplifier models from OpAmp.lib for other simulators?

csrabak, 2024-01-25:
OpAmps.lib contains some interesting opamp models which are [still?] not available if Spice class simulator is chosen.

Is it a mean to pull these models, or a pointer to the original source so we could update SpiceOpamp.lib?

zergud, 2024-01-25:
May be we can use discussion for such questions?

tomhajjar, 2024-01-26:
Presently there are 2 OpAmp libraries supplied with Qucs-S.
OpAmps.lib -> Qucs
SpiceOpamp.lib -> Qucs-S

I made a new library called OpAmp.lib that should be released eventually. Here is the last update 
OpAmp.lib -> Qucs-S

Back to your question about OpAmps.lib. It was done a very long time ago and optimized to work with Qucs. In general it is worthless for ngspice or LTspice. You should search the LTspice forum and ask questions if you want models not included with LTspice.

[OpAmp library.zip](https://github.com/ra3xdh/qucs_s/files/14059828/OpAmp.library.zip)

ra3xdh, 2024-01-26:
The current implementation of the OpAmp.lib relies on Qucsator netlist syntax features that cannot be back converted to SPICE. It's need either to extend `qucs2spice.cpp` or recreate this library. You can recreate the `OpAmp.lib` library using this manual: https://qucs.sourceforge.net/docs/tutorial/opamp.pdf It explains how the opamp macromodels were done.

ra3xdh, 2024-01-26:
> I made a new library called OpAmp.lib that should be released eventually. Here is the last update

I will add the new models to SpiceOpamp.lib library.

tomhajjar, 2024-01-26:
Vadim

Are you going to replace models in the "new" SpiceOpamp.lib with the versions I did in OpAmp.lib?

Is there a reason for so many versions of "ad822" in the present SpiceOpamp.lib?
 
[image]

ra3xdh, 2024-01-31:
I have added OP491 and LF356 models by @tomhajjar to the SpiceOpamp library. The duplicate AD822 models are removed, because all device were sharing one model. Only AD822 remains in the library.

ra3xdh, 2024-01-31:
Closing as completed after merged #517

ra3xdh, 2024-01-31:
The sent version of the library doesn't contain TLV2371

closed 2024-01-31 by ra3xdh

tomhajjar, 2024-01-31:
I am generating a "replacement" library called OpAmps.lib so the internal labels are the same as the old library.