ra3xdh/qucs_s #508: IGBT thermal project status reporting

tomhajjar, 2024-01-29:
Having issues getting an IGBT thermal project fully working. Quite often a transient simulation aborts early.

Should Qucs-S report &#x2713 Simulation successful when this happens? 

[image]

[IGBT_thermal_prj.zip](https://github.com/ra3xdh/qucs_s/files/14089396/IGBT_thermal_prj.zip)

ra3xdh, 2024-01-30:
> Should Qucs-S report Simulation successful when this happens?

I will update error status recognition patterns for Ngspice. This message is not recognized as error, because it has another format.

ra3xdh, 2024-01-30:
Fixed via #511

closed 2024-01-30 by ra3xdh

tomhajjar, 2024-01-30:
Going to be a lot of work getting most IGBT, MOSFET... thermal models to work in ngspice. 

Rohm has a vast array of devices and provides "Spice" and LTSpice versions of "Standard" and "Thermal" models. ngspice doesn't seem to like any of them...