ra3xdh/qucs_s #544: Qucs-S crashes doing S-parameter file analysis

tomhajjar, 2024-02-10:
An old Qucs analysis causes Qucs-S to crash unless I disable the S-parameter analysis of file X1.

[image]

[ATF34143_Amplifier_prj.zip](https://github.com/ra3xdh/qucs_s/files/14228592/ATF34143_Amplifier_prj.zip)

ra3xdh, 2024-02-10:
The reason of the crash was noise data in the s2p file. I have added a fix for this #546 Now Qucs-S+Ngspice will ignore the noise data and give a warning, but the simulation will continue.

closed 2024-02-10 by ra3xdh

tomhajjar, 2024-02-10:
I removed the noise data, deleted and replaced the  s2p and Qucs-S still crashes. 

[ATF34143_Amplifier_prj.zip](https://github.com/ra3xdh/qucs_s/files/14229864/ATF34143_Amplifier_prj.zip)

ra3xdh, 2024-02-10:
Also remove blank lines form the s2p file. Everything is fixed by #546. The fix will be available in the next release.

tomhajjar, 2024-02-10:
I'm doing something wrong...

[image]

[image]

ra3xdh, 2024-02-10:
Please wait for the next release. I will try to prepare a package within the next week.

tomhajjar, 2024-02-10:
Qucs-S didn't like the CR at end of ATF341430_4v_40ma.s2p. Blank line....

ALSO you must select Polar/Rectangular FIRST then choose the file, ATF341430_4v_40ma.s2p. Since majority of s2p files I've seen are polar form, defaulting to polar might be better. 

Alternatively place polar/rectangular first in the menu and make it mandatory choice?

[image]

commit 6e5868ee (2024-02-10): Merge pull request #546 from ra3xdh/fix_544

Fixed #544