ra3xdh/qucs_s #548: Varistor library doesn't work....anymore??

tomhajjar, 2024-02-10:
I haven't used the Varistor library in a long time but when making a test folder of DC sweeping devices, the library didn't work. 

I know it worked in the past #92 (2022) but unsure of the details. I recovered the data from an old backup.

The Curve Tracer project might be useful for new users.

[image]

[image]

[Curve Tracer Testing_prj.zip](https://github.com/ra3xdh/qucs_s/files/14229822/Curve.Tracer.Testing_prj.zip)

ra3xdh, 2024-02-10:
This library doesn't work anymore when using Ngspice-39. If the Ngspice version <= 37 everything works as expected.

ra3xdh, 2024-02-10:
The `qucs2spice::convert_rcl()` need to be modified. Ngspice requires to wrap variable value in braces to distinguish it from model. 

But the best solution will be to remake the Varistor.lib to provide true SPICE netlist. This task cannot be done in the near future, because the library contains about 100 devices.

tomhajjar, 2024-02-10:
I will look into available models.

ra3xdh, 2024-02-10:
I will try to provide a fix of `qucs2spice.cpp` in the next few days.

tomhajjar, 2024-02-12:
The Epcos/TDK, varistor library is for "block" style varistors which have a much higher current rating than the generic leaded disk varistors used in most products. Not really an issue for varistor simulations where voltage is the primary parameter other than the parasitics being higher. Not sure why this type of varistor was used.

[image]

[SIOV_Block_HighE.zip](https://github.com/ra3xdh/qucs_s/files/14235026/SIOV_Block_HighE.zip)

closed 2024-02-12 by ra3xdh