ra3xdh/qucs_s #554: Rewrite documentation

ra3xdh, 2024-02-12:
The documentation on the https://qucs-s-help.readthedocs.io/ is totally outdated now and doesn't reflect the latest application state. This manual need to be 100% rewritten. I am planning to remove the readthedocs repository and replace it by new reference guide. In the long perspective I am planning to supply offline PDF documentation with the release version of the Qucs-S, because readthedocs has limitation on the file size. The documentation must be hosted in a separate repository. This repository must not be integrated in a main source tree. 

I have drafted a contents of the reference guide here: https://github.com/ra3xdh/qucs_s_docs/blob/master/ref_guide/qucs_s_reference.tex Here is a TODO list for documentation update:

- [x] Setup a repository containing LaTeX sources of the documentation: https://github.com/ra3xdh/qucs_s_docs/
- [ ] Prepare Qucs-S simulation reference guide
- [ ] Deprecate repository on the readthedocs and replace links in the Qucs-S GUI. 

The documentation writing consumes much more time than writing a code. I have no time to complete this task in the near year. The contribution is highly welcome here. Here are some articles on Qucs-S that could be used as the starting point. Sorry, it is available in Russian only:

* Simulation guide: https://habr.com/ru/articles/789452/
* RF simulation guide: https://habr.com/ru/articles/735508/
* Qucs-S overview: https://habr.com/ru/articles/678526/ and https://habr.com/ru/articles/778666/

tomhajjar, 2024-02-12:
I recovered/re-created and updated a number of the schematics in the Help section. There are numerous errors in the transformer section. Changes to ngspice and Qucs-S will require schematics and text to change.

Ask for schematics prior to updating the text. It could save a of of time and frustration...

ra3xdh, 2024-02-12:
> Ask for schematics prior to updating the text

Thanks. I will request the schematics if necessary.  

For convenience of not LaTeX users here is the draft of the new reference guide. The structure of the new manuals differs significant from the readthedocs version.
 
[qucs_s_reference.pdf](https://github.com/ra3xdh/qucs_s/files/14241590/qucs_s_reference.pdf)

viticleri, 2024-02-12:
Hi,
Let's see, there is a lot of work to be done in terms of documentation. I've checked some of your articles in Russian (very good, by the way!) and it's really a lot of work to translate them, read them chronologically, take advantage of what's not outdated right now and with that, be able to make a good document updated, which is didactic and clear, which does not create confusion and which at the same time is not boring. 

In the end, when you study a subject in depth, the best way to understand it is to work on it and make good documentation that serves to explain to others the subject you have supposedly learned.

3813127458, 2024-02-16:
I could join to help writing manual sections using LaTeX.
I just need some guidance how you set up version control and peer review of the manual. 

There is a potential solution for online real-time collaborative LaTeX writing:
https://github.com/overleaf/overleaf
https://www.overleaf.com/home-2

In the manual I would like to see also a section about circuit optimization. 
QUCS-S does probably not support optimization right now, but I use ASCO for this purpose quite successful with the QUCS-S netlist for ASCO: https://asco.sourceforge.net/

ra3xdh, 2024-02-25:
> There is a potential solution for online real-time collaborative LaTeX writing:

Yes, I am aware about Overleaf. But I am not sure if overleaf supports Git repository integration. The documentation should be put under version control. 

> In the manual I would like to see also a section about circuit optimization.

The optimization is currently supported only using Qucsator. Ngspice in principle supports ASCO, but for Qucs-S optimization is possible only if you export netlist and run it from the CLI. The ASCO support for Qucs-S+Ngspice could be added in the future versions.

3813127458, 2024-02-25:
Yes, Overleaf supports Git repository integration:

https://www.overleaf.com/learn/how-to/Git_integration
https://www.overleaf.com/learn/how-to/Overleaf_premium_features
https://www.overleaf.com/user/subscription/plans

But is is a "premium feature" not for free. Therefore, I think in our case Overleaf is probably useless.

I did write a section 7 about optimization based on your LaTeX template:
[qucs_s_reference.pdf](https://github.com/ra3xdh/qucs_s/files/14397503/qucs_s_reference.pdf)

Here is the related LaTeX source folder:
[QUCS-S.zip](https://github.com/ra3xdh/qucs_s/files/14397508/QUCS-S.zip)

To clean up and to prepare for a large document including IEEE references format, I did the following minor changes in your  LaTeX template:

1. change `\usepackage{hyperref}` to
```
\usepackage[pdftex]{hyperref}
\hypersetup{
	colorlinks=true,
	pdfpagemode=UseNone,
	pdfstartview=FitV,
	linkcolor = black,
	filecolor = black,
	urlcolor = black,
	citecolor = black,
	bookmarksopen,
	bookmarksnumbered,
	bookmarksopenlevel=2,
	pdfmenubar=true,
	pdfwindowui=true
}
```
2. remove `\usepackage{cite}` and replace by 
```
\usepackage[backend=biber,style=ieee]{biblatex}
\addbibresource{./references/references.bib}
```
3. add `\usepackage{float}`
4. add `\usepackage{siunitx}`
5. add lstlisting format:
```
\lstset{
	basicstyle=\ttfamily,
	columns=fullflexible,
	breaklines=true
}
```
6. add label macros `\fig{}` etc.

The Qucs-S project of the optimization example section 7 is here:
[Example_RLC_Optimization_prj.zip](https://github.com/ra3xdh/qucs_s/files/14397528/Example_RLC_Optimization_prj.zip)

Regarding LaTeX tools I recommend the following setup:

1. Tex Live: https://www.tug.org/texlive/
Advantage: same behavior on Windows and Linux platforms. No utf code page issues etc.

2. TeXstudio: https://www.texstudio.org/
Advantage:  same behavior on Windows and Linux platforms. Very nice LaTeX solution:
Not to forget: the bibliography build setup need to be changed to  `biber` in the settings.

3. Labels, figure filenames and citation references should be a unambiguous integer number. For large documents with hundred of labels, figures and citation references it does not make sense to write meaningful names.  I use a Python script, which posts a integer number into the clipboard, base on the UTC timestamp in micro second resolution:
```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
"""
Created on Sun Feb 25 08:21:54 2024

@author: werner
"""

from datetime import datetime, timezone
import pyperclip

# get current datetime in UTC
utc_dt = datetime.now(timezone.utc)

# convert current datetime in UTC to timestanp integer label
label = str(int(utc_dt.timestamp()*1e6))
print(label)

# copy label string into clipboard
pyperclip.copy(label)
```
Example: `1708889645264020`

4. Sometimes it is necessary to crop PDF figures (print Qucs-S schematic to PDF ) to its contents.
For this purpose I use this Python script:
```

# -*- coding: utf-8 -*-
"""
Created on Tue Nov 20 18:33:35 2018

@author: ARTP2
"""
# This script uses the path of a PDF file in the clipboard and performs:
# 1: crop the PDF to its content (to remove border)
# 2: export EMF for power point (vector graphics resolution)
# 3: export SVG for Inkscape editing
#
# Prerequisites: Inkscape (for emf and svg export) and TeXLive (for pdfcrop) must be installed

import pyperclip
import subprocess
from pathlib import Path
import sys
import os

# copy file path to clipboard
def path_copy():
    txt = pyperclip.paste()
    if sys.platform == 'linux':
        fpath = txt[txt.find('file://')+7:-1]
        print(txt)
        return txt
    elif sys.platform.startswith('win'):
        print(txt)
        return txt
    else:
        print('Error: no linux or windows OS detected')
        return -1

pf = Path(path_copy())
print(pf)

# crop PDF

cmdCommand = "pdfcrop " + str(pf) + " " + str(pf)
print (cmdCommand)
subprocess.call(cmdCommand, shell=True)
print ('done')

# pop up PDF
if os.name == 'posix':
    cmdCommand = 'okular ' + str(pf.with_suffix('.pdf'))
    subprocess.call(cmdCommand, shell=True)
else:   
    os.startfile(pf.with_suffix('.pdf'))

## PDF to EMF
#cmdCommand = r"C:\PROGRA~1\Inkscape\inkscape.exe -f=" + txt + r" --export-emf=" + head + "\\" + fname[0] + ".emf"
#print(cmdCommand)
#process = subprocess.Popen(cmdCommand.split(), stdout=subprocess.PIPE)
#output, error = process.communicate()
#print(output)
#
## PDF to EMF
#cmdCommand = r"C:\PROGRA~1\Inkscape\inkscape.exe -f=" + txt + r" --export-plain-svg=" + head + "\\" + fname[0] + ".svg"
#print(cmdCommand)
#process = subprocess.Popen(cmdCommand.split(), stdout=subprocess.PIPE)
#output, error = process.communicate()
#print(output)
```

I am not sure how my proposals are in line with the work flow of your team.
If I should contribute to a Git repository including version control, someone need to guide me how to setup the Git flow.

ra3xdh, 2024-02-26:
> If I should contribute to a Git repository including version control, someone need to guide me how to setup the Git flow.

The usage of Git is sufficient easy especially on Linux platform. You don't need programming knowledge to use Git. The following instruction is suitable  for all repository: https://docs.github.com/en/get-started/exploring-projects-on-github/contributing-to-a-project

In a short words you should make the following:
* Make fork
* Clone the the LaTeX sources from fork to your hard drive
* Make new branch
* Make changes and commit
* Push changes to your fork
* Open PR
* Delete branch after PR is accepted; update `master` branch on your local copy

ra3xdh, 2024-02-26:
Please follow the following rules while making PRs to documentation:

* Don't put under version control auto-generated files like `*.aux`
* The putting of the large binary files (>= 1Mb) under version control should be avoided

paulmcquad, 2024-02-27:
I was wondering would it be better to change to Asciidoctor. Change Latex to Asciidoctor is easier and more organized. it outputs to different formats like pdf, html5 etc.

* See website for more details:  https://asciidoctor.org/ 

P.S. I could help start to process by Forking and adding a PR.

ra3xdh, 2024-03-10:
@3813127458 I have created an empty Overleaf project for the future documentation: https://www.overleaf.com/project/65ed52fbd51e0623f11e285a You write everything here, and we can synchronize with Github later. Please tell me how can I invite you for editing?

3813127458, 2024-03-14:
@ra3xdh:  I did create my Overleaf account according your invitation and everything works seamless.
I did upload and compile my LaTeX contribution: a sub-section about optimization.
Probably now you can review the result.

We can say now: yes, Overleaf works.
However, I am not sure if Overleaf is optimum solution. I never used it before.
I know that students use Overleaf to work on larger documents together as a group. 
It is a real time system. Everybody can see changes in real time.
Disadvantage: cloud handling latency and uploads.

In general I prefer a fast workflow (in this case: Texlive, Texstudio, Inkscape, Gimp, Python, Octave) on my local computer and to upload my final result just once.

So, if you and the other potential contributors agree, I would continue to set up the Git flow on a Linux platform.

I need to write a git-based script to perform the following steps, which you did mention above:

* Make fork
* Clone the the LaTeX sources from fork to your hard drive
* Make new branch
* Make changes and commit
* Push changes to your fork
* Open PR
* Delete branch after PR is accepted; update `master` branch on your local copy

In case you have a solution for such kind of script, please post it.

ra3xdh, 2024-03-25:
@3813127458 

> I am not sure if Overleaf is optimum solution.

Overleaf has one big advantage. It may attract non-Git users and Windows users for writing documentation. I prefer offline workflow too, but let's try to use Overleaf for documentation preparation.

3813127458, 2024-03-25:
OK I agree, let's go this way.

ra3xdh, 2024-07-14:
I have added Russian version of the reference guide: https://github.com/ra3xdh/qucs_s_docs/tree/master/ref_guide_RUS I have started to collect the information from my articles about Qucs-S here.

k4kfh, 2025-01-22:
Hi, I'm interested in helping with documentation. I'm learning to use the QUCS-S so I am in a good spot to contribute I think, at least at the introductory documentation level.

Is this still being actively worked on? Can somebody share a current status update?

ra3xdh, 2025-01-22:
There is no activity on the documentation in the recent months. The best way of contribution is to fork the repository: https://github.com/ra3xdh/qucs_s_docs/ , make changes at some chapter and send me a pull request.

tomhajjar, 2025-01-22:
@k4kfh 

Most of the examples in the help section are available. Many are in the Discussion section. Having the schematics used in examples and subcircuits used to make the Libraries helps when writing the accompanying text.

k4kfh, 2025-01-22:
> [@k4kfh](https://github.com/k4kfh)
> 
> Most of the examples in the help section are available. Many are in the Discussion section. Having the schematics used in examples and subcircuits used to make the Libraries helps when writing the accompanying text.

Thank you, great tip!

k4kfh, 2025-01-22:
Ah, I see. That makes sense.

I don't think there is a limit on total repository size for public GitHub repositories, so the Sphinx site could be as big as needed. The limitation would be the PDF size (since it would be a GitHub Actions "artifact"), and the number of GitHub actions compute "minutes" used every month. GitHub gives 500MB "artifact" storage, and 2000 compute "minutes" per month, for all public repositories.

So maybe it makes the most sense to continue to use ReadTheDocs for hosting Sphinx (due to simplicity), and simply use GitHub Actions to generate the PDF from the Sphinx files? It could be set up to only generate the PDF when you make a Release on GitHub, which would probably keep within the limits of the free tier.

Let me know if that sounds reasonable to you and I can set up a proof-of-concept.

ra3xdh, 2025-01-22:
> So maybe it makes the most sense to continue to use ReadTheDocs for hosting Sphinx 

ReadTheDocs didn't host the Sphinx sources of the old Qucs-S documentation. The sources were hosted here: https://github.com/ra3xdh/qucs_s-help It was uploaded to ReadTheDocs using the webhook.

k4kfh, 2025-01-22:
> ReadTheDocs didn't host the Sphinx sources of the old Qucs-S documentation. The sources were hosted here: https://github.com/ra3xdh/qucs_s-help It was uploaded to ReadTheDocs using the webhook.

Sorry, what I meant was "maybe it makes the most sense to continue to use ReadTheDocs for hosting the resulting Sphinx site, while storing the Sphinx sources in version control on GitHub. I understand the sources for the existing ReadTheDocs site were stored in Git.

Here is a proposal:
* Add a ``user-docs`` subdirectory to this repository, and store Sphinx sources there. This allows us to trigger documentation-related actions whenever you make a Release on this repo.
* Set up ReadTheDocs using the webhook, pointing to ``ra3xdh/qucs_s/user-docs`` for sources.
* Set up a GitHub Action that runs when you create a GitHub Release in ``ra3xdh/qucs_s``, which generates the offline PDF documentation. This would make offline PDF docs available for download at the same place as the software binaries themselves.

What are your thoughts on that method?

ra3xdh, 2025-01-22:
> Set up ReadTheDocs using the webhook, pointing to ra3xdh/qucs_s/user-docs for sources.

Why to use ReadTheDocs instead of Github pages: https://coderefinery.github.io/documentation/gh_workflow/ What is advantage of ReadTheDocs? I remember the ReadTHeDocs has a 2Mb limit in the past and QUcs-S documentation exceeded this limit. Also I don't own the ReadTheDocs account, and the person who has setup the ReadTheDocs for Qucs-S is not responding.

k4kfh, 2025-01-22:
> Why to use ReadTheDocs instead of Github pages: https://coderefinery.github.io/documentation/gh_workflow/ What is advantage of ReadTheDocs?

I think the biggest advantage is less maintenance. ReadTheDocs handles all the maintenance of the Sphinx build system, we would not need to mess around with updating Sphinx versions, which commands to use within the GitHub Actions workflow, etc. ReadTheDocs "just works".

Another advantage is that GitHub Actions could end up costing money depending on how often the Sphinx builds occur, and how many compute minutes they take. ReadTheDocs Community plan doesn't have those limitations in place, so we can build documentation at every commit without worrying about hitting paywalls.

> I remember the ReadTHeDocs has a 2Mb limit in the past and QUcs-S documentation exceeded this limit.

I'm not really sure about this, I can't find anything on their site about it. I'm going to send their support an email and ask for clarification.

If there truly is a 2MB limit, then I agree with you that ReadTheDocs is not a good option and we should build Sphinx using GitHub Actions/GitHub Pages.

> Also I don't own the ReadTheDocs account, and the person who has setup the ReadTheDocs for Qucs-S is not responding.

Another good point. I guess we could make another account, but that doesn't really solve the long-term problem of all the power being with one person. I will also ask ReadTheDocs Support about this, maybe there is a way to give ownership to multiple people. If not, I guess that would be another benefit of GitHub Actions/GitHub Pages.

---

I'll report back when I hear from ReadTheDocs support.

k4kfh, 2025-01-22:
# Answers from ReadTheDocs Support

## Storage Limitations

**Question:**
> Is there any kind of storage limitation? Our documentation will be very rich with images, and another developer said he thought you had a 2MB storage limit on the Community plan. I can't find any information confirming/denying that on your website. Can you clarify?

**Answer:**
> There are no hard limitations, but generally around 5GB is when we start worrying about it. I've put a PR together to note this explicitly: https://github.com/readthedocs/readthedocs.org/pull/11943

## Share Access with Multiple People

**Question:**
> We want to share access to the ReadTheDocs settings with multiple people. That way if one developer is unavailable, the other developers can still control the ReadTheDocs settings. Is that possible with the Community plan? We are hosting with GitHub so it would be best to tie access to GitHub accounts. 

**Answer:**
> Yes, you can create multiple accounts with maintainer access: https://docs.readthedocs.io/en/stable/guides/managing-maintainers.html

---

The only thing I can think of that might get over 5GB of storage would be very large image files, and if it became a big problem, we could use a GitHub Actions script to check image sizes and compress if needed.

I looked at the Maintainers feature and that looks like it would solve the access problem. We might need to make a fresh account this one time, but as long as the key contributors all have maintainer access to that account, I would think that solves any future lockout issues.

@ra3xdh What are your thoughts?

k4kfh, 2025-01-27:
> if you are agreed to collect the existing documentation and translate it to Sphinx.

Sure, that sounds good to me.

> I can create a repository for Sphinx documentation and add you as collaborator

Can I suggest making the documentation a subfolder in this main repository? I can fork and set up a demo of that before we do anything permanent. But doing it in this repository instead of a separate one will use the features of ReadTheDocs more effectively I think. For example, we can automatically build docs for each pull-request in this repo, so documentation updates can be reviewed at the same time as code updates. It also makes it possible to keep version-specific documentation, and automatically include offline documentation when you make a Release in this repository.

If you don't want to do that I'm also happy to do it in a separate repository.

ra3xdh, 2025-01-27:
> If you don't want to do that I'm also happy to do it in a separate repository.

I can create a separate repository and add it later as Git submodule to the main repository.

k4kfh, 2025-01-27:
That works too. If you add me as a collaborator on the Git repo, I can set up the new ReadTheDocs account and add you as a maintainer.

ra3xdh, 2025-02-01:
@k4kfh I have restored access to the old ReadTheDocs account https://qucs-s-help.readthedocs.io/ I have just added you as a collaborator to https://github.com/ra3xdh/qucs_s-help Accept this invitation. Please tell me your actual email address to add you to the maintainers list in ReadTheDocs. The ReadTheDocs doesn't support adding by Github account. My actual email could be found in my Github profile. 

You are free to wipe the old documentation in `master` branch and replace it with new. The documentation in `qucs_s-help` repository is 100% outdated and only mislead user. I have archived an old documentation in `legacy` branch:  Don't modify this branch: https://github.com/ra3xdh/qucs_s-help/tree/legacy

k4kfh, 2025-02-03:
@ra3xdh Sounds great! I've accepted the GitHub Collaborator invite and I sent you an email from my email address, so you'll have my contact info for ReadTheDocs.

ra3xdh, 2025-04-05:
This issue may be closed. The rewrite of the documentation using Sphinx is in progress. The discussion on the documentation features may be continued in the individual issues in `qucs-s-help` repository: https://github.com/ra3xdh/qucs_s-help/issues

closed 2025-04-05 by ra3xdh