ra3xdh/qucs_s #560: Getting a .FUNC formula to produce data to plot

tomhajjar, 2024-02-13:
In converting a @dwarning VBIC project to Qucs-S I ran into a few issues. The final one is I cannot get "NF" in the Function below to generate data to plot. I am missing a step. In addition Qucs-S won't let you enter a ",".

.FUNC NF(inoise,rsource) = 10*log10(inoise^2/(4*1.38e-23*300*rsource))

[image]

[VBIC_Analysis_prj.zip](https://github.com/ra3xdh/qucs_s/files/14255431/VBIC_Analysis_prj.zip)

ra3xdh, 2024-02-14:
There may be Regexpt error in the input field. But I would totally deprecate the .FUNC device because of low demand. This device duplicates the functions of the Include script. As the workaround of this bug you may use the INCLUDE SCRIPT device and enter whatever you want. 
[image]

ra3xdh, 2024-02-14:
.FUNC has effect only before the simulation. You have to add Nutmeg equation referencing the .FUNC This information could be found in the Ngspice manual. Everything works as expected after applying corrections. 
[image]

closed 2024-02-14 by ra3xdh

tomhajjar, 2024-02-14:
I made the changes and replicated @dwarning data. 

Wasn't able to get the INCLUDE SCRIPT to work. ngspice error messages are not helpful

As you stated .FUNC is extremely rare. Other than @dwarning, I could only find 3 ngspice examples.

[image]

[image]

[VBIC_Analysis_prj.zip](https://github.com/ra3xdh/qucs_s/files/14283628/VBIC_Analysis_prj.zip)

ra3xdh, 2024-02-14:
You should write the full statement in the INCLUDE script. Add the .FUNC keyword. Unfortunately the error messages in Ngspice are very cryptic and often don't help to find the error.

dwarning, 2024-02-14:
Please inspect the netlist. Look like nf(... stands direct in netlist and with n in first position a Verilog-A model is expected, Something goes wrong in netlisting.

tomhajjar, 2024-02-14:
The other issue is the 2nd .INCLUDE script (INCLSCR2) isn't disabled.