ra3xdh/qucs_s #562: Qucs-s closes when closing the tuner with several parameters

Tyskanchik, 2024-02-14:
Windows10

When I try to reset values or close tuner with several parameters, program closes.
If there is only one parameter, then program works.

[qucs-s_win10_issue.webm](https://github.com/ra3xdh/qucs_s/assets/70572897/f8427fff-3190-4fa4-bc46-efd2e25c7315)
[wireless_charge.zip](https://github.com/ra3xdh/qucs_s/files/14276339/wireless_charge.zip)

ra3xdh, 2024-02-14:
I cannot reproduce the crash using the latest build and Linux version.  Only Windows version is affected.

zergud, 2024-02-14:
```
Debug: Tuner::slotElementValueUpdated() (C:/Users/zergud/CLionProjects/qucs_s/qucs/dialogs/tuner.cpp:760, void TunerDialog::slotElementValueUpdated())
Warning: QObject::connect: No such signal QProcess::error(QProcess::ProcessError) in C:/Users/zergud/CLionProjects/qucs_s/qucs/extsimkernels/abstractspicekernel.cpp:72 (:0, )
Warning: QObject::connect: No such signal QProcess::error(QProcess::ProcessError) in C:/Users/zergud/CLionProjects/qucs_s/qucs/extsimkernels/abstractspicekernel.cpp:72 (:0, )
Debug: "C:\\Users\\zergud\\.qucs\\spice4qucs" (C:/Users/zergud/CLionProjects/qucs_s/qucs/extsimkernels/ngspice.cpp:431, virtual void Ngspice::slotSimulate())
Fatal: ASSERT failure in QList::at: "index out of range", file C:/msys64/mingw64/include/qt6/QtCore/qlist.h, line 431 (C:/msys64/mingw64/include/qt6/QtCore/qlist.h:431, )
```

closed 2024-02-16 by ra3xdh