ra3xdh/qucs_s #567: Qucs-S user library portability

tomhajjar, 2024-02-15:
I noticed that the two RF resistor examples "Test_chip_res_basic.sch" and "Test_RATF_EquivCircD.sch" don't work anymore. The devices used in these examples work in the 7 other examples in the attached project. 

https://github.com/ra3xdh/qucs_s/tree/current/examples/ngspice/RFLumpComp

One issue is the examples you uploaded point to your user_lib folder. I didn't dig too deeply into the other issues but it got me thinking how to make user created projects and libraries "portable". LTspice allows placing everything required into a single folder. Is there a way other than making subcircuits containing the symbol information to place a user created Qucs compatible library in the project folder?

[Resistor_RF-Models_prj.zip](https://github.com/ra3xdh/qucs_s/files/14298924/Resistor_RF-Models_prj.zip)

ra3xdh, 2024-02-15:
> Is there a way other than making subcircuits containing the symbol information to place a user created Qucs compatible library in the project folder?

There is no way to place library at the project directory. Library devices support specifying the full path to libraries, but library manager will look only at system library or user_lib directories. This was already reported and related to #4. It still be unclear how to introduce library search paths and improve the library portability.

ra3xdh, 2024-04-07:
Relative paths implemented with #640. I will implement the visibility of the libraries in project directory myself.

ra3xdh, 2024-04-07:
The #676 implements showing the libraries from the project directory in the library tree.
[image]

closed 2024-04-07 by ra3xdh