ra3xdh/qucs_s #575: Qucs current source changed

tomhajjar, 2024-02-17:
I was converting from Qucs formulas to Nutmeg and the original Qucs current source doesn't work anymore for values of "1A" or "1 A". It has to be "1". It has to be a recent change.

[image]

[Impedance Measurements_prj.zip](https://github.com/ra3xdh/qucs_s/files/14317456/Impedance.Measurements_prj.zip)

ra3xdh, 2024-02-17:
I didn't change anything in the current sources. The `A` means atto and Ngspice is not case sensistive, but Qucsator is case sensistive.

tomhajjar, 2024-02-17:
Two things changed. 

Polarity is flipped from 0.0.24 -> 24.1+ and "A" now means "Atto". 

I am next going to testing ngspice versions.

[image]

ra3xdh, 2024-02-17:
The polarity was fixed by #246 It is correct now.

tomhajjar, 2024-02-17:
The change in "A" happened in ngspice 42

ra3xdh, 2024-02-17:
I have just added a fix for amperes unit recognition https://github.com/ra3xdh/qucs_s/commit/92d9a39cb04ed9b18fbabf272e9375401f1af477 Closing as completed.

closed 2024-02-17 by ra3xdh