ra3xdh/qucs_s #587: windows height

dwarning, 2024-02-18:
Compiling latest current branch on Ubuntu 22.04 (old laptop) follows in very large heights for all simulation windows. This was not the case in previous builds:
[image]

iwbnwif, 2024-02-18:
From the greying of the main window it looks like "Attach Modal Dialogs" might be selected in Gnome. It still doesn't explain the resizing, but it may be a side effect.

If you have Gnome Tweaks installed, you can try disabling it there under "Windows".

dwarning, 2024-02-21:
I tried this Gnome setting configuration - but no change. The dialog boxes have still maximum height.
Behaviour has changed since 24.0.

ra3xdh, 2024-02-21:
I never observed this issue using KDE+X11 or Windows. I changed nothing related to dialog size policy.

iwbnwif, 2024-02-21:
Did you try to resize the window? The simulation window size is now retained from the previous time since #529. 

I am using Ubuntu 22.04 and don't see this problem, but it may be that if Attach Modal Dialogs setting is active, then the initial (default with no configuration in the settings file) may somehow set the height to maximum. 

I will look into it a bit later, but in the meantime please can you try adjusting the size and seeing if it is retained between components.

dwarning, 2024-02-21:
Yes - it is the retaining effect. I can resize, close the program, reopen and see same size as before.
Sorry, I never noticed this feature before.
Thank you - I will close.

closed 2024-02-21 by dwarning