ra3xdh/qucs_s #592: Configure picks Qt6 over Qt5 when both arfe available

yurivict, 2024-02-18:
Version: 24.1.0
FreeBSD 14

ra3xdh, 2024-02-18:
This is done intentionally. You have to set the Qt version explicitly if both are installed. See #217 I am not sure what option may help for FreeBSD.

yurivict, 2024-02-19:
@ra3xdh 

There used to be the option ```WITH_QT6```, but now there is no such option because it was removed.
```QT_DEFAULT_MAJOR_VERSION``` is set, but it doesn't affect the Qt choice.

Before #217 it was possible to choose the Qt generation, and now it doesn't appear to be possible.

This is ok, I can patch CMakeLists.txt, but this is still a regression IMO.

ra3xdh, 2024-02-19:
It is not a regression. It is done intentionally. No need to patch CMakeList. Qt version could be selected using environment variables. The reasons to remove `WITH_QT6` were described by @dsm in #217

yurivict, 2024-02-19:
FYI:  ```QT_DEFAULT_MAJOR_VERSION``` doesn't seem to work due to an apparent bug in cmake: https://gitlab.kitware.com/cmake/cmake/-/issues/25691

yurivict, 2024-02-19:
@yurivict 

According to the answer to the [cmake bug report](https://gitlab.kitware.com/cmake/cmake/-/issues/25691) that I created [this line](https://github.com/ra3xdh/qucs_s/blob/current/CMakeLists.txt#L36) is a mistake, and it has nothing to do with Qt selection in cmake. It only affects some subsequent Qt-related macros. There doesn't appear to be any environment variable for this either.

Could you please consider adding back WITH_QT6, because https://github.com/ra3xdh/qucs_s/pull/217 had no clear explanation why it should have been removed from @dsm .

ra3xdh, 2024-02-20:
Yes, I will revert the explicit Qt6 selection using `WITH_QT6` Also I am planning to switch the build to Qt6 as default and drop Qt5 support till the end of this year. So the issue will be self-fixed in the future.

ra3xdh, 2024-03-06:
I have reverted WITH_QT6 flag in #629. @yurivict Please check everything on FreeBSD.

closed 2024-03-06 by ra3xdh