ra3xdh/qucs_s #598: Does present version of qucs-s deprecate qucsator?

csrabak, 2024-02-19:
In the simulator paths and the selection drop menu qucsator appears as one of the supported simulators.

However it does not come with the distribution of presente qucs-s release (at least not in the Windows binary).

Is it expected users keep older qucs installation and point to the binary there or do we another solution?

ra3xdh, 2024-02-19:
Qucsator is supported, but you have to compile it yourself. You can also take it from old 0.0.19 Windows release. Qucsator team doesn't provide binaries. This issue will be resolved with introduction of Qucsator_RF #370 Please wait or take Qucsator from old release, or compile it yourself.

closed 2024-02-19 by ra3xdh

csrabak, 2024-02-19:
OK! Thanks.