ra3xdh/qucs_s #600: Example works 24.1.0 doesn't work 24.1.0+

tomhajjar, 2024-02-20:
The wingspread example stopped working using the last nightly you sent me.

[image]

ra3xdh, 2024-02-20:
The Nutmeg equations were changed in the recent version #513 . The attaching of the nutmeg equation to DC sweep was fixed by #595. Now the DC means DC OP, and SW1 means DC sweep. You should select the simulation type from the drop-down list in the equation properties. We are planning to prepare correction release 24.1.1 in the next two weeks. Please wait. You may set `Simulation=ALL` in the equation properties as a workaround.

ra3xdh, 2024-02-20:
Closing because already fixed by #595

ra3xdh, 2024-02-25:
The wingspread schematic example updated by #603