ra3xdh/qucs_s #602: Lossless TLine netlist is wrong in some cases

tomhajjar, 2024-02-20:
The TLine netlist has unneeded entries confusing ngspice.

T1 1 2 3 4 Z0=50 Td=1ns -> good  - Unfortunately Qucs will add "F=0"/"F=" and "NL=0"/"NL="

T1 1 2 3 4 Z0=50 F=1e9 NL=0.25 -> good - Unfortunately Qucs will add "Td=0"/"Td="

T1 1 2 3 4 Z0=50 Td=1ns F=0 NL=0 -> bad - If Td used, F and NL should not be in netlist

T1 1 2 3 4 Z0=50 Td=0 F=1e9 NL=0.25 -> bad - Td=0 will nullify F=1e9 NL=0.25

T1 1 2 3 4 Z0=50 Td=1ns F= NL= -> bad

[image]

[image]

[Transmission_Line_prj.zip](https://github.com/ra3xdh/qucs_s/files/14352029/Transmission_Line_prj.zip)

dwarning, 2024-02-21:
The attached patch file will solve the problems in tline parameter handling.
All examples show reasonable results.

[Tline.patch.txt](https://github.com/ra3xdh/qucs_s/files/14363630/Tline.patch.txt)

ra3xdh, 2024-02-24:
@dwarning Thanks for providing a patch. I have added your path as a part of #610

closed 2024-02-24 by ra3xdh

dwarning, 2024-02-24:
@ra3xdh Thank you.
But as mentioned in #601 the patch is only a workaround. qucs-s gui has to work as user of ngspice who take care for the both specification modes of tline: 1. Specify Td (F and Nl omitted) 2. Specify F and Nl (Td is omitted and not(!) given in netlist)

tomhajjar, 2024-02-24:
1. Specify Td (F and Nl omitted) 
2. Specify F and Nl (Td is omitted and not(!) given in netlist)
3. Specify ONLY F (Td AND NL is omitted and not(!) given in netlist) and ngspice -> NL=0.25

tomhajjar, 2024-04-10:
I revisited my project above and ngspice TLine is now working with ver 24.2.0.