ra3xdh/qucs_s #605: Compare foundry measurements with model performances

dwarning, 2024-02-21:
I use s-parameter files to compare foundry measurements with model performances. Question is, how can I use s-parameter from different operating points in parametric simulations only in plotting. Tried to put them in one file doesn't work (with and without an empty line between the data blocks). Using for each file an extra 2-port setup? Looks a bit elaborate.

_Originally posted by @dwarning in https://github.com/ra3xdh/qucs_s/discussions/475#discussioncomment-8450011_

dwarning, 2024-02-26:
@ra3xdh : Could you comment please, Thanks.

ra3xdh, 2024-02-27:
I didn't fully understand what are trying to do. Could you submit a schematic? The s2p implementation for Ngspice is taken froms2spice utiliy: https://groups.io/g/LTspice/topic/75745954?p=Created,,,20,2,0,0 It may have limitations. If the s2p features for Qucs-S+Ngspice are not sufficient you may use Qucs-S+Qucsator. Qucsator has native support of touchstone files.

dwarning, 2024-02-27:
Thanks.
The problem is not to generate or to convert the s-param files. I have them for different operating points each. But I am not able to show them all in compare to the parametric simulation sweep.

[image]

Only one curve (red) from one file is shown here. The other not. I want only show the data - not simulate with them.

I can provide schematic/project, but you have to include IHP model lib.

onehebe, 2024-03-08:
Hi dwarning,
As far as I can see, the standard  frequency-based snp file doesn't contains bias/op information.
If multiple blocks of data with different bias/op included in one file, the parser would terminate at the end of first block.
Maybe you can upload your sample snp file. 
We will see how to improve the snp parser function then.

dwarning, 2024-03-08:
Attached are the 3 sp files as shown before in my picture. I greped them from IHP's mdl file.
[spar-vbe_078_vcb_025.s2p.txt](https://github.com/ra3xdh/qucs_s/files/14535491/spar-vbe_078_vcb_025.s2p.txt)
[spar-vbe_088_vcb_025.s2p.txt](https://github.com/ra3xdh/qucs_s/files/14535492/spar-vbe_088_vcb_025.s2p.txt)
[spar-vbe_096_vcb_025.s2p.txt](https://github.com/ra3xdh/qucs_s/files/14535493/spar-vbe_096_vcb_025.s2p.txt)