ra3xdh/qucs_s #607: RF component library

tomhajjar, 2024-02-22:
Will the RF component library re-appear once qucsator-RF is available? #370

I tested the library a long time ago and found things that could be improved. Also some devices work differently in QucsStudio.

[image]

[image]

ra3xdh, 2024-02-24:
There is no explicit group RF devices in the Qucs-S. All of the mentioned devices are available in the "lumped devices" group. In principle I can ad this group in the next release. Two devices: spiral inductor and loop inductor should be backported from Qucs-0.0.20.
[image]

ra3xdh, 2024-02-24:
>  Also some devices work differently in QucsStudio

I cannot comment anything on this, because QucsStudio has a closed source. I have no wish to synchronize Qucs-S and closed source projects like QucsStudio.

ra3xdh, 2024-02-24:
I have added a TODO list for this issue.

tomhajjar, 2024-02-24:
I had issues when I ported a Qucs project to QucsStudio. Amplifier uses power gain versus voltage gain. I think hybrid ports are different as well. One issue with hybrid is you couldn't tell how to connect the ports. The "normal" way to connect doesn't work. It would also help if the ports are labeled especially when rotated or flipped.

Hybrid and Coupler can both be configured to be 90/180 and the connections are different. 

[image]

[RF_Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/14391954/RF_Test_prj.zip)

ra3xdh, 2024-02-24:
Fixed by #610.

ra3xdh, 2024-02-24:
[image]

closed 2024-02-24 by ra3xdh

tomhajjar, 2024-02-24:
Need nightly build for testing...

wsteffe, 2024-03-09:
Hi @ra3xdh, 
I have just pulled the github repo and rebuilt qucs-s but I do not have the RF components shown in your post.
[image]

ra3xdh, 2024-03-09:
@wsteffe Everything is correct. The most of RF devices are Qucsator-only. Select Qucsator from the drop-down list and you will see the full list. This is done to prevent insertion of the incompatible device in schematic. You may have to compile Qucsator from source depending on your platfrom.

wsteffe, 2024-03-10:
Hi @ra3xdh, Thanks for explanation.
It would be better if  RF devices were redefined using behavioural models or equations that can be processed by ngspice. But I do not know if it is possible to do that.

ra3xdh, 2024-03-10:
> It would be better if RF devices were redefined using behavioural models or equations

It is not possible for the most of RF devices. See #94

PR #610 (2024-02-24): Add loop and spiral inductors
Fixes #607 

* backported loop and spiral inductors from Qucsator
* aded RF devices group
* Some devices moved from lumped to RF devices